cesanta/mongoose #1129: Best way to support "Single Page Application" ?

TomzBench, 2020-07-15:
I need to configure the server to redirect all 404 to the root html page so that 404 is handled by the client side application.  What happens currently is that if I refresh the page I receive a 404 error, however, the URL is valid and the root application will know how to handle the url.

I have tried modifying the server_opts.url_rewrites configuration with 404=/path/to/my/root. However this does not seem to work.

What is the best way to support a single page web application?  Should i manually serve the files?

TomzBench, 2020-07-15:
I am currently doing something like this 

```
static void
process_file(http_s* http, struct mg_connection* c, struct http_message* m)
{
    char b[512];
    const char* root = http->serve_opts.document_root;
    int err;
    unsigned int _p;
    struct stat st;
    struct mg_str _s, _u, _h, path, _q, _f;
    err = mg_parse_uri(m->uri, &_s, &_u, &_h, &_p, &path, &_q, &_f);
    if (err) return;
    snprintf(b, sizeof(b), "%s%.*s", root, path.len, path.p);
    err = stat(b, &st);
    if (err) {
        log_debug("(HTTP) [%s] not found... Serving root", b);
        snprintf(b, sizeof(b), "%s%s", root, "/index.html");
        mg_http_serve_file(c, m, b, mg_mk_str("text/html"), mg_mk_str(""));
    } else {
        log_debug("(HTTP) [%s] found... ", b);
        mg_serve_http(c, m, http->serve_opts);
    }
}
```
If the file is found I call mg_serve_http as normal, and if the file is not found I serve index.html (the root of my SPA.  Because simple url rewrites won't seem to let me handle the case where I want to serve assets/js/css like normal and only redirect on the 404.
But is there a better way?

jeremypoulter, 2020-07-16:
Have you considered using the anchor part of the URL rather than the path? eg instead of http://example.com/some/path use http://example.com/#some/path. That way the server is not involved at and all the logic stays client side.

cpq, 2020-12-25:
What @jeremypoulter said.
The best way to handle SPA is to use URI fragments.

Have mongoose serve static files from some directory, and dynamic api calls tie to the `/api/...` prefix.
Then, have a client side (e.g. preact-based) to handle fragment paths.

Example: https://github.com/cesanta/mongoose/tree/master/examples/complete

closed 2020-12-25 by cpq