cesanta/mongoose #1132: Long time wait until connection closed if http response with more than one chunk and without content-length header.

charlesso, 2020-07-22:
I found that response with more than one chunk and without content-length header will take a long time to completed. 

everything working fine after adding this fix. 

```
  again:
    req_len = mg_parse_http(io->buf, io->len, hm, is_req);
    if (req_len > 0 && (pd == NULL || pd->finished)) {
      /* New request - new proto data */
      pd = mg_http_create_proto_data(nc);
      pd->rcvd = io->len;
    }
    // *** fix here ***
    else{
      /* following chunks */
      pd->rcvd += io->len; 
    }
    // *** fix done ***
  
    if (req_len > 0 &&
        (s = mg_get_http_header(hm, "Transfer-Encoding")) != NULL &&
        mg_vcasecmp(s, "chunked") == 0) {
      mg_handle_chunked(nc, hm, io->buf + req_len, io->len - req_len);
    }
```

tzpeng, 2020-08-01:
I have the same problem, but mine didn’t work

rojer, 2020-08-01:
does your mongoose.c have [this fix](https://github.com/cesanta/mongoose/commit/95fcb261ebb44a9664c336824f0968d56faaa50b)?

tzpeng, 2020-08-01:
@rojer Thanks !

charlesso, 2020-08-03:
Edit previous comment for finding pd->rcvd need to increase in case of MG_HTTP_CONTENT_LENGTH_UNKNOWN only.

tianyJC, 2020-08-04:
thanks, effective fix

cpq, 2021-03-08:
Chunked API, refactored, please see
https://github.com/cesanta/mongoose/tree/master/examples/http-streaming-client

closed 2021-03-08 by cpq