cesanta/mongoose #1135: A buffer overflow error in mg_get_http_header

BushraAloraini, 2020-08-12:
A buffer overflow error in mg_get_http_header function in mongoose/src/mg_http.c in Mongoose 6.18, where header_names and header_values have a bound of (MG_MAX_HTTP_HEADERS); however, there is no check to ensure that the loop does not exceed the upper bound. A Maliciously crafted  http header can trigger this bug. To fix it, modify the loop condition into:
   
 for (i = 0;i < MG_MAX_HTTP_HEADERS && hm->header_names[i].len > 0  ; i++)

rojer, 2020-08-19:
fix sent, thank you!

closed 2020-08-19 by cpq

rojer, 2020-08-20:
actually, this will not happen in practice because mg_http_parse_headers will not populate the last slot:
https://github.com/cesanta/mongoose/blob/86f391c70158a0ee553746e326bc996a3a39b4c0/src/mg_http.c#L408

but it's good to have a range limit too.

BushraAloraini, 2020-08-20:
Thank you, rojer, for your response.  This bug has been detected using a static analysis tool that is part of a research study. Hence, your input is very important. Actually, the static analysis tool was run with an older version of mongoose, which was used with an Android app (AdAway); at that version, mg_get_http_header is defined in mongoose.c. I just realized that mg_get_http_header is still defined in mongoose.c with the newest version (I thought the definition was moved to a new file). I see that mg_get_http_header that is defined in mongoose.c has been fixed, which is great. Also, I see that mg_get_http_header has been called 22 times in mongoose.c file.  Hence, there are many execution paths, so some might trigger the error.

cpq, 2020-09-19:
@BushraAloraini so you ran an analysis tool that spotted a false positive, and logged a https://cve.mitre.org/cgi-bin/cvename.cgi?name=CVE-2020-25756 ?