cesanta/mongoose #1142: NULL pointer dereference caused by incorrect error handling of calloc in mg_send_dns_query (mongoose.c:11542)

cve-reporting, 2020-08-27:
Incorrect handling of the value returned by calloc may lead to NULL pointer dereference in case of an allocation error.

Vulnerable code (mongoose.c:3464-3465):
>   struct mg_dns_message *msg =
>       (struct mg_dns_message *) MG_CALLOC(1, sizeof(*msg));
>   struct mbuf pkt;
>   struct mg_dns_resource_record *rr = &msg->questions[0];
>   DBG(("%s %d", name, query_type));
>   mbuf_init(&pkt, 64 /* Start small, it'll grow as needed. */);
>   msg->transaction_id = ++mg_dns_tid;

See following recommendations for details (especially the calloc example):
https://wiki.sei.cmu.edu/confluence/display/c/ERR33-C.+Detect+and+handle+standard+library+errors

The issue can be reproduced and tested using ErrorSanitizer (https://github.com/Samsung/ErrorSanitizer).

Reproduction steps:
1. Download and compile ErrorSanitizer (https://github.com/Samsung/ErrorSanitizer#compilation)
2. Set ESAN_PATH to ErrorSanitizer directory
3. Download and unzip attached map esan_map_004

4. Compile Mongoose examples
5. Enable coredumps using: 
>     ulimit -c unlimited
6. Run tested application with ErrorSanitizer using:
> AFL_USE_STDIO=Y LD_PRELOAD=${ESAN_PATH}/error_sanitizer_preload.so ./examples/coap_client/coap_client < esan_map_004
7. Analyze crash using:
> gdb ./examples/coap_client/coap_client core

You should receive similar output:

> GNU gdb (Ubuntu 7.11.1-0ubuntu1~16.5) 7.11.1
> Copyright (C) 2016 Free Software Foundation, Inc.
...
> [New LWP 20868]
> [Thread debugging using libthread_db enabled]
> Using host libthread_db library "/lib/x86_64-linux-gnu/libthread_db.so.1".
> Core was generated by `./examples/coap_client/coap_client'.
> Program terminated with signal SIGSEGV, Segmentation fault.
> #0  0x000000000041b126 in mg_send_dns_query (nc=0x2154d00, name=0x21548c0 "coap.me", query_type=1) at ../../mongoose.c:11551
> 11551	  msg->transaction_id = ++mg_dns_tid;

8. You can obtain more information using 'backtrace' and 'backtrace full' commands:

> (gdb) backtrace
> #0  0x000000000041b126 in mg_send_dns_query (nc=0x2154d00, name=0x21548c0 "coap.me", query_type=1) at ../../mongoose.c:11551
> #1  0x000000000041be37 in mg_resolve_async_eh (nc=0x2154d00, ev=2, data=0x7ffffe40d0f4) at ../../mongoose.c:11998
> #2  0x0000000000407e84 in mg_call (nc=0x2154d00, ev_handler=0x41bce9 <mg_resolve_async_eh>, user_data=0x21548c0, ev=2, ev_data=0x7ffffe40d0f4) at ../../mongoose.c:2288
> #3  0x0000000000409abb in mg_if_connect_cb (nc=0x2154d00, err=0) at ../../mongoose.c:3014
> #4  0x000000000040b635 in mg_mgr_handle_conn (nc=0x2154d00, fd_flags=2, now=1598538479.227843) at ../../mongoose.c:3874
> #5  0x000000000040c0c8 in mg_socket_if_poll (iface=0x2153480, timeout_ms=1000000) at ../../mongoose.c:4082
> #6  0x0000000000408764 in mg_mgr_poll (m=0x7ffffe40d400, timeout_ms=1000000) at ../../mongoose.c:2493
> #7  0x00000000004021dd in main (argc=1, argv=0x7ffffe40d528) at coap_client.c:73

> (gdb) backtrace full
> #0  0x000000000041b126 in mg_send_dns_query (nc=0x2154d00, name=0x21548c0 "coap.me", query_type=1) at ../../mongoose.c:11551
>         msg = 0x0
>         pkt = {buf = 0x0, len = 0, size = 0}
>         rr = 0x20
> #1  0x000000000041be37 in mg_resolve_async_eh (nc=0x2154d00, ev=2, data=0x7ffffe40d0f4) at ../../mongoose.c:11998
>         now = 1598538479
>         req = 0x21548c0
>         msg = 0x0
>         user_data = 0x21548c0
> #2  0x0000000000407e84 in mg_call (nc=0x2154d00, ev_handler=0x41bce9 <mg_resolve_async_eh>, user_data=0x21548c0, ev=2, ev_data=0x7ffffe40d0f4) at ../../mongoose.c:2288
>         flags_before = 2
> #3  0x0000000000409abb in mg_if_connect_cb (nc=0x2154d00, err=0) at ../../mongoose.c:3014
> No locals.
> #4  0x000000000040b635 in mg_mgr_handle_conn (nc=0x2154d00, fd_flags=2, now=1598538479.227843) at ../../mongoose.c:3874
>         err = 0
>         worth_logging = 1
> #5  0x000000000040c0c8 in mg_socket_if_poll (iface=0x2153480, timeout_ms=1000000) at ../../mongoose.c:4082
>         fd_flags = 2
>         mgr = 0x7ffffe40d400
>         now = 1598538479.227843
>         min_timer = 0
>         nc = 0x2154d00
>         tmp = 0x0
>         tv = {tv_sec = 999, tv_usec = 999998}
>         read_set = {fds_bits = {0 <repeats 16 times>}}
>         write_set = {fds_bits = {64, 0 <repeats 15 times>}}
>         err_set = {fds_bits = {0 <repeats 16 times>}}
>         max_fd = 6
>         num_fds = 1
>         num_ev = 1
>         num_timers = 0
>         try_dup = 1
> #6  0x0000000000408764 in mg_mgr_poll (m=0x7ffffe40d400, timeout_ms=1000000) at ../../mongoose.c:2493
>         i = 0
>         num_calls_before = 0
> #7  0x00000000004021dd in main (argc=1, argv=0x7ffffe40d528) at coap_client.c:73
>         mgr = {active_connections = 0x2154d00, hexdump_file = 0x0, ctl = {4, 3}, user_data = 0x0, num_ifaces = 1, num_calls = 0, ifaces = 0x2153010, nameserver = 0x0}
>         nc = 0x2153540
>         address = 0x41d2e8 "udp://coap.me:5683"
>

cpq, 2020-12-18:
Closing this, as the DNS codepath has been refactored since 7.0

closed 2020-12-18 by cpq

cve-reporting, 2021-01-04:
Hi @cpq !
But you do confirm that previous versions are affected?

cpq, 2021-01-04:
We do not maintain previous versions.