cesanta/mongoose #1144: Multiple Buffer over-read issues can happen due to improper length checks in function mbuf_insert

BushraAloraini, 2020-08-30:
Multiple Buffer over-read issues can happen due to improper length checks in function mbuf_insert. The issues occur within memcpy, and memmove function calls, which are controlled by tainted data, and there is no check to ensure that the length (the number of bytes) does not exceed the actual length of the source buffer.

Here are the lines of code that include the issues: memcpy line 1456 and 1475 and memmove line 1454 and 1473
[image]
  

Sometimes buffer over-read can be hard to discover, thus to see how this issue happens, a printing statement that shows the values can be used such as
`fprintf(stdout, "Debug_INFO: mbuf_insert  memmove line1454: actual len of source ( a->buf +off ) = %ld \n" , strlen(a->buf + off));  //  <--- added for debugging`

Thus, add the following printing statements to the source code just before the mentioned unsafe statements that show the number of bytes to be copied from the source and the actual length of the source buffer:
  

[image]

 
Then compile Mongoose code, and then compile and run captive_dns_server in the example directory.

In another terminal, write the following command: `dig @localhost -p 5533 +padding=511`

[image]

 
**Note** that padding option controls the number of bytes to be copied (len) in memcpy in line 1475
So, the actual length of the source buffer in memcpy in line 1475 is 4, but the program is trying to read 511 that is controlled remotely.

And try other options like `dig @localhost -p 5533 www.google.com +bufsize=5000  +padding=511 A`

So what happens when specifying the number of bytes to be copied larger the actual buffer is that the program reads adjacent memory, as in the Heartbleed bug, and this is critical as it can b triggered by maliciously crafted inputs.
The issue is related to: https://vuldb.com/?id.119690  -- CVE-2018-10945 

 
How to fix this issue? Ensure that the length of bytes (the third argument) does not exceed the actual source buffer(the second argument). One assumption in this code is that the buffer in _a->buf_ has a length of _a->len_ which is not always true! This is also applied to _buf_ and _len_.

cpq, 2020-08-30:
Thank you, appreciated.
An existing `test/Makefile`:`fuzz` target already checks the `mg_parse_http()` function. Do you think the `fuzz.c` could be extended to include the mbuf issues you've discovered?

BushraAloraini, 2020-08-30:
Possibly. The memcpy function call in line 1475 is triggered by (mg_recv_udp->mbuf_append->mbuf_insert)  path. Note that the default behavior of dig DNS query is to use UDP; **however, the packet query can be sent using TCP**. Hence it is worth testing mg_recv_tcp as well. Other function calls mentioned in this report, such as memmove in line 1454 can be triggered by multiple paths such as:

- mg_dns_copy_questions -> mbuf_append ->mbuf_insert
- mg_dns_encode_record -> mg_dns_encode_name -> mg_dns_encode_name_s -> mbuf_append ->mbuf_insert

Thus, mbuf_insert function can be triggered by multiple paths. Therefore, it is better to have the safety check just before the unsafe function calls.  I discovered this problem using BEFinder tool, a security static analysis tool that I developed as part of a research project, and it shows that these functions are receiving tainted input without checking the length of the actual buffer, which may lead to buffer over-read. Fuzzing can work as well, yet fuzzing may be less effective when dealing with buffer overread and information leakage since it might not cause a program crash or undefined behavior.

attritionorg, 2020-08-31:
@BushraAloraini Did you use ASAN? If so could you post the full output for one of the instances please?

cpq, 2020-12-18:
Closing this.

The mbuf API has been refactored in 7.0,
The CI is using ASAN, and mongoose is integrated into the asan-enabled oss-fuzz.

closed 2020-12-18 by cpq