cesanta/mongoose #1148: mongoose upload file don't support chunked

niuzhenbo, 2020-09-16:
when i use 
 * void ev_handler(struct mg_connection *nc, int ev, void *ev_data) {
 *   switch (ev) {
 *     ...
 *     case MG_EV_HTTP_PART_BEGIN:
 *     case MG_EV_HTTP_PART_DATA:
 *     case MG_EV_HTTP_PART_END:
 *       mg_file_upload_handler(nc, ev, ev_data, upload_fname);
 *       break;
 *   }
 * }
handle upload file , The received file is  corrupt. because every time the file is written, the length before the block is written into the file

[image]

[image]

xingyun86, 2020-09-25:
can you receive MG_EV_HTTP_PART_BEGIN message? I test it but I only receive ev=100.

xingyun86, 2020-09-27:
I solve it with myself now . Also thanks very much.

cpq, 2020-12-23:
Please use the latest code , which provides an alternative upload mechanism:
https://github.com/cesanta/mongoose/tree/master/examples/file-upload

closed 2020-12-23 by cpq