cesanta/mongoose #1155: what's mean is     (void) timeout_ms;     ?

moll33er, 2020-10-09:
`static time_t mg_null_if_poll(struct mg_iface *iface, int timeout_ms) {
  struct mg_mgr *mgr = iface->mgr;
  struct mg_connection *nc, *tmp;
  double now = mg_time();
  /* We basically just run timers and poll. */
  for (nc = mgr->active_connections; nc != NULL; nc = tmp) {
    tmp = nc->next;
    mg_if_poll(nc, now);
  }
  (void) timeout_ms;     //what does this line do?
  return (time_t) now;
}`
(void) timeout_ms;  //what does this line do?

lnksz, 2020-11-02:
Please learn to use the tripple back-quotes for formatted code-block ```
```c
static time_t mg_null_if_poll(struct mg_iface *iface, int timeout_ms) {
    struct mg_mgr *mgr = iface->mgr;
    struct mg_connection *nc, *tmp;
    double now = mg_time();
    /* We basically just run timers and poll. */
    for (nc = mgr->active_connections; nc != NULL; nc = tmp) {
        tmp = nc->next;
        mg_if_poll(nc, now);
    }
    (void) timeout_ms; //what does this line do?
    return (time_t) now;
}
```

In C casting a function parameter to `(void)` is to mark it "unused". See https://stackoverflow.com/a/3599170/6232305

moll33er, 2020-11-04:
Thank you for your answer! I should learn more!

closed 2020-11-04 by nliviu