cesanta/mongoose #1163: Add support for wasm mime type

voidah, 2020-11-24:
Hello,

I'm getting this error message in web developper console on chrome because text/plain mime type is sent by mongoose for web assembly file (not tested on other browser): `wasm streaming compile failed: TypeError: Failed to execute 'compile' on 'WebAssembly': Incorrect response MIME type. Expected 'application/wasm'.`

Here is how I fixed it:

```
diff --git a/mongoose.c b/mongoose.c
index ebabcc0..42f7fee 100644
--- a/mongoose.c
+++ b/mongoose.c
@@ -5982,6 +5982,7 @@ static const struct {
     MIME_ENTRY("asf", "video/x-ms-asf"),
     MIME_ENTRY("avi", "video/x-msvideo"),
     MIME_ENTRY("bmp", "image/bmp"),
+    MIME_ENTRY("wasm", "application/wasm"),
     {NULL, 0, NULL},
 };
```

What do you think? Thanks!

bagbell, 2020-12-03:
works on Edge for me. Thanks!

paladin-t, 2020-12-15:
I second the motion.

cpq, 2020-12-15:
Integrated, thank you.

https://github.com/cesanta/mongoose/commit/f176e1ae598c8ca008af4c3f9810614009e332a8

closed 2020-12-15 by cpq