cesanta/mongoose #1164: mongoose causes a redraw of the window [MFC]

TwilightShaw, 2020-11-26:
**Goal**
I have created a dialog box that is hidden by default.
When I fetch data with mongoose, the timeout will show the dialog box.
**Question**
When I fetched the data with the following code, it caused **a redraw of the window** and the window showed up directly.
```
        struct mg_mgr mgr;
	struct mg_connection *nc;
	mg_mgr_init(&mgr, NULL);
	nc = mg_connect_http(&mgr, ev_value_handler, url.c_str(), header.c_str(), NULL);
	mg_set_protocol_http_websocket(nc);
	// Timer
	mg_set_timer(nc, mg_time() + time_out_second);
	while (s_exit_flag == 0) {
		mg_mgr_poll(&mgr, 100);
	}
	mg_mgr_free(&mgr);
```
Breakpoint 1:  mg_set_timer(nc, mg_time() + time_out_second); 
                        Window status: hide
Breakpoint 2: g_mgr_free(&mgr);
                        Window status: show 
I'm just starting to learn how to use mongoose and I want to know why this is happening.

PiterYask, 2020-11-27:
The mangoose timer works only for active connection. If the connection is closed the timer will not work and case MF_EV_TIMER will not be called.

cpq, 2020-12-13:
Please use the latest version - it has a different timer mechanism.

closed 2020-12-13 by cpq