cesanta/mongoose #1165: Problem with OpenSSL and the non-blocking sockets

marcelwirtz, 2020-11-30:
Hey,

we had problems with SSL handshakes and the non-blocking mode on sockets with Mongoose Version up from 6.7. The problem occurs if there are two requests send right behind each other. We experienced it mostly happened on POST- / PUT-requests, which result in a curl 56 error.

After a bit of research we could fix the problem with the following loop inside the mg_ssl_if_handshake function.

```
...
res = server_side ? SSL_accept(ctx->ssl) : SSL_connect(ctx->ssl);
  while(res == -1) {
      fd_set fds;
      FD_ZERO(&fds);
      FD_SET(nc->sock, &fds);

      switch(SSL_get_error(ctx->ssl, res)) {
          case SSL_ERROR_WANT_READ:
              select(nc->sock, &fds, NULL, NULL, NULL);
              break;
          case SSL_ERROR_WANT_WRITE:
              select(nc->sock, NULL, &fds, NULL, NULL);
              break;
          default: abort();
      }
      res = server_side ? SSL_accept(ctx->ssl) : SSL_connect(ctx->ssl);
  }
...
```

This is not the best solution because it's making mongoose blocking again, but the problem with non-blocking sockets is, that the SSL_accept / SSL_connect function is only called one time, if the socket is not read- or writable it fires the SSL_ERROR_WANT_READ or SSL_ERROR_WANT_WRITE error.

Maybe you guys can switch Mongoose to use OpenSSL memory BIOs, or find another thing that will perform better since only fire the error and close the connection.

Thanks and kind regards,
Marcel

Further information:
[https://www.openssl.org/docs/man1.1.0/man3/SSL_accept.html](https://www.openssl.org/docs/man1.1.0/man3/SSL_accept.html)
[OpenSSL example using memory BIO with non-blocking socket IO ](https://gist.github.com/darrenjs/4645f115d10aa4b5cebf57483ec82eca)

cpq, 2020-11-30:
Thank you.

Does it happen on the latest 6.18 ?

marcelwirtz, 2020-11-30:
yes, it happens also with version 6.18

cpq, 2020-12-07:
We're working on a 7.0 release - could you check out the master branch and try it on the latest codebase, please?
The TLS functionality has been refactored.

marcelwirtz, 2020-12-07:
Since the basic structure changes and our C ++ wrapper no longer works so easily with this version. Unfortunately, I can't just test the whole thing, I have to convert our wrapper beforehand. Since it is not a persistent problem either, it probably makes sense to test it with the software with which it occurs.

cpq, 2020-12-11:
@marcelwirtz  could you come up with the minimal reproducible example please?
We could integrate that into the CI unit test, so you'll be covered forever.

cpq, 2020-12-24:
The 7.0 has been released.
Current codebase does not have the blocking snippet in the SSL handshake function.
Can you reproduce an issue now?

cpq, 2021-01-25:
No feedback, closing this.

closed 2021-01-25 by cpq