cesanta/mongoose #1172: WebSocket

carlos1995pinto, 2020-12-12:
This is not a problem but a suggestion.
I think that  a function (specially in server side) that sends data through WebSocket as soon as it is available should me made.  What i mean is, when function **mg_ws_send()** is called it should send the content of **_buf_** input argument to the other "side" of the WebSocket, instead of appending that content to a buffer. 
In conclusion, i think that current **mg_ws_send()**  function should be maintained as it is but another one with the purpose mentioned above should be created.

cpq, 2020-12-12:
Do I understand it correctly,
your suggestion is to modify the logic of the sending function so that it tries to push data immediately, and only if send fails or is incomplete, store the remaining data in the output buffer and drain it later?

carlos1995pinto, 2020-12-13:
Not modify, but create a new one with the purpose of push data immediately. If it fails to do that, i think it should return an error to the user. It should not store the remaining data (data not sent) in the output buffer.

cpq, 2020-12-13:
A competing API, plus incomplete partial writes which user should handle?
It looks like the benefit/price ratio is too high for this. I am not sure there is any benefit at all.
Closing this - please free to reopen, but provide a clear benefit it provides.

closed 2020-12-13 by cpq