cesanta/mongoose #1177: when i init TLS for HTTPS, I get an HTTP parse error.

bobooboo, 2020-12-18:
To develop HTTPS api, I used your example 'http-restful-server' source code. 
and I just added below code for TLS (HTTPS)

```c
// added code for tls
struct mg_tls_opts tls_opt;
tls_opt.cert = crt;
tls_opt.certkey = key;
mg_tls_init(conn, &tls_opt);
```
and
```
// build options
-DMG_ENABLE_OPENSSL=1 -DMG_ENABLE_LOG=1 -DMG_ENABLE_HTTP_DEBUG_ENDPOINT=1
```
SO, 
```c
#include "mongoose.h"

// change "http://..." to "https://..."
static const char *s_listen_on = "https://localhost";
static const char *s_web_directory = ".";

// This RESTful server implements the following endpoints:
//   /api/f1 - respond with JSON string {"result": 123}
//   /api/f2/:id - wildcard example, respond with JSON string {"result": "URI"}
//   any other URI serves static files from s_web_directory
static void fn(struct mg_connection *c, int ev, void *ev_data, void *fn_data) {
  if (ev == MG_EV_HTTP_MSG) {
    struct mg_http_message *hm = (struct mg_http_message *) ev_data;
    if (mg_http_match_uri(hm, "/api/f1")) {
      mg_http_reply(c, 200, "", "{\"result\": %d}\n", 123);  // Serve REST
    } else if (mg_http_match_uri(hm, "/api/f2/*")) {
      mg_http_reply(c, 200, "", "{\"result\": \"%.*s\"}\n", (int) hm->uri.len,
                    hm->uri.ptr);
    } else {
      mg_http_serve_dir(c, ev_data, s_web_directory);  // Serve static files
    }
  }
  (void) fn_data;
}

int main(void) {
  struct mg_mgr mgr;                            // Event manager

  // ADD
  struct mg_tls_opts tls_opt;
  struct mg_connection* conn;
  tls_opt.cert = crt;
  tls_opt.certkey = key;

  mg_mgr_init(&mgr);                            // Initialise event manager
  conn = mg_http_listen(&mgr, s_listen_on, fn, NULL);  // Create HTTP listener

  // ADD
  mg_tls_init(conn, &tls_opt);

  for (;;) mg_mgr_poll(&mgr, 1000);             // Infinite event loop
  mg_mgr_free(&mgr);
  return 0;
}
```
If you run this code, you can get the following debug message.
```
2020-12-18 06:09:18  E mongoose.c:992:http_cb    0x4 HTTP parse error
2020-12-18 06:09:18  E mongoose.c:2556:accept_co 0x3 accept(3) failed, errno 11
2020-12-18 06:09:18  E mongoose.c:2556:accept_co 0x3 accept(3) failed, errno 11
... (repeatly)
2020-12-18 06:09:18  E mongoose.c:2556:accept_co 0x3 accept(3) failed, errno 11
```
c->recv.buf in http_cb(...), a handler that processes http packets, cannot catch any HTTP packets.
and print LOG  "HTTP parse error".

my environment is 
```
ubuntu: 20.08 LTS or 12.08 LTS
cmake: 3.16.3 or 2.8
curl : 7.68
openssl: 1.1.1
```

i used testing for CURL
```
curl --cacert cacert.pem --cert ./test.pem --key ./test.key -X POST https://localhost -d "111" -v --insecure
```

I used openssl to test if the certificate is OK, and it works fine.
```
openssl s_server -key ./test.key -cert ./test.pem -CAfile catest.pem -accept 4433 -msg
```

Could you tell me the problem of this code?
Or please write an HTTPS example code.

Thanks.
ps. Congratulations on the release of mongoose version 7.0.

cpq, 2020-12-18:
Thank you.

http-restful-server updated to support TLS/SSL

closed 2020-12-18 by cpq

PerfectSlayer, 2020-12-23:
Thanks for the example.

As @bobooboo I thought we must initiative tls at main.

Another question about tls usage with v7 is, when should we do use `mg_tls_free` and should we use it all? Thanks.

cpq, 2020-12-23:
@PerfectSlayer the only public TLS API is `tls_init()`. All other TLS functions are called by Mongoose core, they are not intended for the outside.

PerfectSlayer, 2020-12-23:
Thanks for the explanation. I was surprised I found no doc about the other function.