cesanta/mongoose #1181: how use http client send json data in mongoose release7.0

GitHubjiangxl, 2020-12-29:


GitHubjiangxl, 2020-12-29:
// Print HTTP response and signal that we're done
static void client_fn(struct mg_connection *c, int ev, void *ev_data, void *fn_data) 
{
    if (ev == MG_EV_CONNECT) 
    {
        // Connected to server
        if (mg_url_is_ssl(notice_url)) 
        {
            // If notice_url is https://, tell client connection to use TLS
            struct mg_tls_opts opts = {.ca = "ca.pem",.cert = "client.crt", .certkey = "client.key"};
            mg_tls_init(c, &opts);
        }
        // Send request
        // log_debug(0,"uri:%s,%s", mg_url_uri(notice_url),(char*)fn_data);
        // mg_printf(c, "GET %s HTTP/1.0\r\n\r\n", mg_url_uri(notice_url));
        mg_printf(c, "POST %s HTTP/1.0\r\n\"Content-Type: application/json\"\r\n\r\n%s\r\n", mg_url_uri(notice_url), (char*)fn_data);
    } 
    else if (ev == MG_EV_HTTP_MSG) 
    {
        // Response is received. Print it
        struct mg_http_message *hm = (struct mg_http_message *) ev_data;
        log_debug(0, "%.*s", (int) hm->message.len, hm->message.ptr);
        c->is_closing = 1;         // Tell mongoose to close this connection
        *(bool *) fn_data = true;  // Tell event loop to stop
    }
}

>>>
mg_printf(c, "POST %s HTTP/1.0\r\n\"Content-Type: application/json\"\r\n\r\n%s\r\n", mg_url_uri(notice_url),  this have problem?

cpq, 2020-12-29:
Please publish the snippet that creates client connection, and specifically, what is passed as `fn_data`.

GitHubjiangxl, 2020-12-30:
Item item;
        char *str =  cJSON_Print(root);
        log_debug(0, "%s\n", strlen(str) >= sizeof(item.data) ? "有损数据传输":"完整数据传输");
        memset(item.data, 0x00, sizeof(item.data));
        strncpy(item.data, str, strlen(str) < sizeof(item.data) ? strlen(str) : sizeof(item.data)-1);
        free(str);
        int ret = put_item(&item);
        if (ret != 0)
        {
            log_error(0, "put_item failed");
        }

item.data is json string.this is fn_data.

cpq, 2020-12-30:
The snippet you've posted is not what has been asked for.
If you don't invest into issue report, there is no reason for investing into the answer.
Closing this.

closed 2020-12-30 by cpq