cesanta/mongoose #1182: Documentation for previous releases

mill1000, 2021-01-01:
Reading the web-based docs at https://cesanta.com/docs/ is definitely one of the best ways to understand how to use Mongoose. However, it only contains the latest releases. We aren't necessarily all ready to make the jump to v7.0.

Please make the web-based documentation available for previous releases. Jumping through the markdown in the git repository is much more painful.

cpq, 2021-01-02:
Mongoose updates are very often fix security holes, therefore staying with the older version is very discouraged.
We won't provide rendered docs for the previous releases.
Please stay updated.

closed 2021-01-02 by cpq

mill1000, 2021-01-07:
I'd feel better about staying updated if the 7.0 wasn't broken and feature incomplete.

cpq, 2021-01-07:
@mill1000 could you elaborate on your last statement, please?

mill1000, 2021-01-07:
It seems there were a few issues (#1187, # 1183, #1176 , #1173) with 7.0, which were swiftly resolved. That's fantastic, however these fixes are upstream of the actual release.

If I'm actively developing and affected by the above issues, I now have to build against master or some random commit. Assuming my build environment allows me to do that and I'm not locked into specific packages. 

Unfortunately, I now discover that `mg_connect_http` is crippled compared to previous releases, so my efforts to port my development from 6.18 to 7.0 are halted.

So at this point, I choose to step back to 6.18, where I know I can get the job done but alas you've stripped the rendered documentation for 6.18 so I'm forced to trawl through the repo to find the answers I need.

I just think it would be more user friendly to provide access to prior release documentation for a period of time when you make major changes to the library.

cpq, 2021-01-07:
@mill1000 , thank you

7.0 is a major release, it breaks the API, which is expected from a major release.

6.x is API is not perfect and not set in stone, therefore if there is a change, it is not necessarily a step back. This project started in 2004 and there were many API changes on the way - however most basic concepts have been stabilised.

Staying on a previous release is fine.

Said that, we are trying to be responsive, therefore feel free to file issues. So if you don't mind, please move to 7.0 and report your issues - we'll handle them in a timely manner.

Appreciate your comments, they are useful.