cesanta/mongoose #1183: 7.0 SSI support?

krame505, 2021-01-02:
After putting some considerable effort into upgrading my project to 7.0, I discovered that HTTP server-side includes do not seem to work anymore.  Is this intentional?  

The release notes at https://github.com/cesanta/mongoose/releases/tag/7.0 state that "Porting from the 6.x version is straightforward, as all concepts stayed intact."  This is highly misleading if features were indeed removed in 7.0.

cpq, 2021-01-02:
The CGI functionality is gone for good, while SSI is not.
Just pushed https://github.com/cesanta/mongoose/commit/c7e09bd7b4e684837bc13292f324dff62fc0cfa7 which adds include file and include virtual.
The `exec` and `call` directives are not implemented and probably won't be, cause `exec` is insecure, and `call` is discouraging the commonly adopted practice of creating web apps (e.g. react + json endpoints).

closed 2021-01-02 by cpq

cpq, 2021-01-02:
Updated release notes to mention CGI support removal.