cesanta/mongoose #1188: 6.xx to 7.0 upgrade documentation

mill1000, 2021-01-07:
v7.0 completely changed the API. Are there any plans to provide documentation on how best to upgrade?

e.g. The things I've encountered
- `mg_register_http_endpoint` is no more. What's the correct method to mimic this behavior? Should I just call `mg_http_listen` for each of these endpoints?
- `resp_code` and `resp_status_msg` are gone. How do I check HTTP responses? Do I have to manually parse the headers now? The http-client examples does not demonstrate anything useful regarding HTTP responses.
- Looks like `mg_send_head` + `mg_printf` became `mg_http_reply`. I like this. `mg_send_response_line` also got wrapped into `mg_http_reply`.
- `mg_sock_addr_to_str` became `mg_ntoa`...or did it become `mg_straddr`? There's no documentation on `mg_straddr`?
- `MG_F_SEND_AND_CLOSE` became `c->is_closing`? This is not clear. Am I commanding the connection to close or is it telling me it's closing?
- `MG_EV_SEND` became `MG_EV_WRITE`.
- The old `mg_connect_http` took arguments for the header and body data. The new `mg_http_connect` does not. Do I have to monitor for `MG_EV_ACCEPT` and manually generate my request?

My gut feeling is that this update is a major step **backward** for the this product.

cpq, 2021-01-07:
## mg_register_http_endpoint

Multiple endpoints are implemented in the callback:

```c
static void cb(struct mg_connection *c, int ev, void *ev_data, void *fn_data) {
  if (ev == MG_EV_HTTP_MSG) {
    struct mg_http_message *hm = (struct mg_http_message *) ev_data;
    if (mg_http_match_uri(hm, "/api/config/get")) {
        ...  // url1
    } else if (mg_http_match_uri(hm, "/api/config/set")) {
        ...   // url2 , and so on
    } else {
      // Everything else
      struct mg_http_serve_opts opts = {.root_dir = "web_root"};
      mg_http_serve_dir(c, ev_data, &opts);
    }
  }
}

...
mg_http_listen(&mgr, url, cb, arg);

```

## resp_code

The `hm->uri` holds the response code

## mg_ntoa() and mg_aton()

They are now correctly handle IPv6 addresses

## MG_F_SEND_AND_CLOSE

`MG_F_SEND_AND_CLOSE ` is `c->is_draining`
`MG_F_CLOSE_IMMEDIATELY ` is `c->is_closing`

## MG_EV_SEND

That's correct

## mg_connect_http

This one is actually feels like a step backward - we'll review the API more carefully and might change the signature

Thanks for the feedback.
Won't comment on the "backward" part.

cpq, 2021-01-25:
No more feedback - closing this.
Thank you.

closed 2021-01-25 by cpq

mill1000, 2022-01-01:
Coming back to this after a year and here are some additional thoughts on the new API.

## Lack of `mg_register_http_endpoint`
The lack of `mg_register_http_endpoint` is a bummer. Recreating the behavior in the callback is annoying to me.

### First, an endless sea of ifs and else ifs.
I don't like this. I can't imagine many do.
```c
if (ev == MG_EV_HTTP_MSG) 
  {
    struct mg_http_message* hm = (struct mg_http_message*) ev_data;
    if (mg_http_match_uri(hm, "/api1"))
      // Handle API1
    else if (mg_http_match_uri(hm, "/api2"))
      // Handle API2
    else if (mg_http_match_uri(hm, ...))
      // Sure hope I don't have a lot of these...
    else if (mg_http_match_uri(hm, ...))
      // And so on...
    else 
      // ....
  }
```

### Second, passing fn_data to specific URIs is now a PITA.
We used to be able to do this.
```c
void http_start(int some_arg)
{
  struct mg_mgr manager;
  mg_mgr_init(&manager, NULL);

  // Create listening connection
  struct mg_connection* connection = mg_bind(&manager, "80", httpEventHandler, nullptr);

  // Register our API endpoints with specific data
  UserData api1;
  mg_register_http_endpoint(connection, "/api1", apiEventHandler, &api1);

  UserData api2(some_arg);
  mg_register_http_endpoint(connection, "/api2", apiEventHandler, &api2);
  
  while(1)
    mg_mgr_poll(&manager, 10);
}
```
Nice. Clean. Easy to pass arguments from the caller.

But now we have to either:
**A. Create the fn_data _within_ the callback. That's gross, and god forbid if we want to pass an argument to it.**
```c
static void cb(struct mg_connection* c, int ev, void* ev_data, void* fn_data) 
{
  static UserData api1;
  static UserData api2; // TODO WTF how do we get an argument here now?

  if (ev == MG_EV_HTTP_MSG)
  {
    struct mg_http_message* hm = (struct mg_http_message*) ev_data;

    if (mg_http_match_uri(hm, "/api1"))
      apiEventHandler(c, ev, ev_data, &api1);
    else if (mg_http_match_uri(hm, "/api2"))
      apiEventHandler(c, ev, ev_data, &api2);
    else 
      httpEventHandler(n, ev, ev_data, nullptr)
  }
}

void http_start(int some_arg)
[... 68 lines cut ...]
### `mg_send_response_line` is gone
There are times when we can't use `mg_http_reply` because we don't know the body contents yet.

And now that `mg_send_response_line` is gone we have to manually `mg_printf` a response...
```c
mg_printf(nc, "HTTP/1.1 %d %s\r\n", status_code, mg_http_status_code_str(status_code));
```
but that won't actually work because `mg_http_status_code_str` is inaccessible.

## Multipart streaming is gone.
It appears multipart streaming is gone. Issues #1110, #1167, #1235, #1246, #1386. This was key to how we handled OTA updates. Don't think we can make use of `mg_http_upload` since we have no filesystem.

## User flags are gone.
The v7 API no longer has user flags (e.g. `MG_F_USER_1`). Not the end of the world, but these were a great way to save some simple state information between events. Based on some examples it appears the `label` field can be used to achieve something similar.

## Response codes.
It would be great if one of the http client examples demonstrated how to access response codes. The current examples just dump the entire response and catch MG_EV_ERROR. From your above comment, plus trial and error I've found that the status code and status message are located in the `uri` and `proto` fields, respectively. Very intuitive.

## Timers need work.
Timers are in need of more work. They are no longer associated with a connection so it's necessary pass a connection or the manager as an argument to perform anything useful (e.g. a timeout, start a connection). This is fine, but there's no way to access the timer object that fired so it's impossible to free the timer or change it's period. I resorted to wrapping them in my own class to make them usable.

**My suggestion would be to change the timer callback signature to take a pointer to the firing timer, and the user argument.**

# Warthog
My solution was to write a simple C++ support library which I've dubbed [warthog](https://github.com/mill1000/warthog). Warthog brings back some of my favorite functions and should make porting from 6.18 to 7.X easier.

kcozens, 2023-09-02:
For the record, another feature that was dropped in v7 is support for the use of #call when using SSI.