cesanta/mongoose #1190: How to implement SSL HTTP sever based on Mongoose

Tiantian930, 2021-01-10:
Hi,
The Mongoose is great and convenient! I can implement a HTTP server by just a few lines. 
However, I have issues about SSL connection. Because recently, I want to upgrade my HTTP server to HTTPS server.

I know Mongoose does support SSL/TLS, but I didn’t find good HTTPS template that I can reuse. 
I am using the latest Mongoos on master branch. I saw there is a C file named **unit_test.c** in the source code package. I saw there is a function named **test_tls()**, looks like it creates a HTTPS server and send a message to itself via SSL. I tried to simulate its work in my code, but I failed to make it work. I post my code snipper below.

My main HTTPS server is quite simple:
[image]

My event handler:
[image]

I disabled the function _**SSL_CTX_set_verify**_ inside the _**mg_tls_init**_, so I can let HTTS server accept the Client connection without client cert validation. 
I wrote a simple client side to send simple string “**Hello! I am Client!**” by **_SSL_write()_** function.

My expectation is: to finish my 3 main jobs inside the **MG_EV_HTTP_MSG** block from my event handle function.

But I found the HTTPS server doesn't work as my expectation. 

The client side showed it successfully sent the **_Hello! I am Client!_** to HTTPS server.
[image]

The HTTS server side showed the connection was established. I saw the mongoose.c showed HTTPS server already received the client request.
[image]

In my original TCP HTTP server implementation, I put all my client request process logic inside the 
**_MG_EV_HTTP_MSG block_**. But I got nothing in my HTTPS server.

My questions:
1. Based on the message, I think the handshake was success and I saw HTTPS server log message **_MG_EV_CONNECT event -> CONNECTION ESTABLISHED_** in **_MG_EV_CONNECT block_**.
2.  But why the client request **_Hello! I am Client!_** didn't appear inside **_MG_EV_HTTP_MSG_** block? If I want to process the request, where should I put my code?
3.  How can I send response back to client via SSL? 
4. I know Mongoose is single thread, is it possible to make it multi-threaded?

I am still a freshman for Mongoose. 
I will very glad if you can provide me a HTTPS server and Client template.

Thanks in advance, I am looking forward for your help!
-Thanks!

cpq, 2021-01-10:
Please see:
- https://github.com/cesanta/mongoose/tree/master/examples/http-restful-server - read the comment on top of the main.c
- https://github.com/cesanta/mongoose/tree/master/examples/multi-threaded

Tiantian930, 2021-01-10:
Thanks! Sergey.
I will look into it for sure.

Actually, I already implemented the multi-threaded HTTP server based on
Mongoose. But I didn't find implemented the HTTPS server in multi-threaded.

I may back to you for more questions.
Thanks for your help!

closed 2021-01-14 by cpq

Tiantian930, 2021-01-14:
Thanks Sergey. I already implemented the HTTPS server with your help!

Thanks. This ticket can be closed.

I am working on how to upgrade it to Multi-Threaded version. I will back to
you if I have any questions.

Thanks.