cesanta/mongoose #1192: How to make DNS parsing timeout？

beyondkyq, 2021-01-13:
When using mg_http_connect. If dns_cb is called during DNS resolution and MG_EV_READ is not triggered, how can dns_cb trigger DNS timeout? In my code, fn_data of dns_cb is always null. Mongoose version is 7.1. Thank you for your answer..

beyondkyq, 2021-01-21:
When I want to make multiple HTTP requests, the code is as follows,I refer to the example of http-client:

``` 
        bool done = false;                        // Event handler flips it to true
 	mg_log_set("4");                          // Set to 0 to disable debug

 	while (true)
 	{
		struct mg_mgr mgr;       
 		done = false;
 		mg_mgr_init(&mgr);                        // Initialise event manager	
 		mg_http_connect(&mgr, s_url, fn, &done);  // Create client connection
		while (!done)
		{
			mg_mgr_poll(&mgr, 1000);    // Infinite event loop
		}
		mg_mgr_free(&mgr);                        // Free resources
 		std::this_thread::sleep_for(std::chrono::milliseconds(1000));
 	}
``` 
If the DNS resolution of URL in mg_http_connect fails once, the following log will appear:
```
2021-01-21 01:49:13    e:\work\code\privatepract 2 -- tchrC
2021-01-21 01:49:13    e:\work\code\privatepract 2 closed
2021-01-21 01:49:13  I e:\work\code\privatepract All connections closed
2021-01-21 01:49:14    e:\work\code\privatepract 1 -> https://test-easy.mall-to.com/api/floor
2021-01-21 01:49:14    e:\work\code\privatepract 2 -> udp://8.8.8.8:53
2021-01-21 01:49:14    e:\work\code\privatepract 2 resolved, sock 00000000000002D4 -> 8.8.8.8:53
2021-01-21 01:49:14    e:\work\code\privatepract 1 resolving test-easy.mall-to.com, txnid 1
2021-01-21 01:49:14    e:\work\code\privatepract 2 tUc 39/39 0
2021-01-21 01:49:15    e:\work\code\privatepract 2 -- tchrc
2021-01-21 01:49:15    e:\work\code\privatepract 1 -- tchRc
2021-01-21 01:49:16    e:\work\code\privatepract 2 -- tchrc
2021-01-21 01:49:16    e:\work\code\privatepract 1 -- tchRc
2021-01-21 01:49:17    e:\work\code\privatepract 2 -- tchrc
2021-01-21 01:49:17    e:\work\code\privatepract 1 -- tchRc
2021-01-21 01:49:18    e:\work\code\privatepract 2 -- tchrc
```
The above logs will loop infinitely. When I debug mongoose.c, I find that mg_mgr_poll will call dns_cb, and ev in dns_cb is MG_EV_POLL, but the parameter fn_data is always null,so the timeout comparison is never performed.It also doesn't trigger MG_EV_CLOSE and MG_EV_ERROR。
If you use an invalid DNS service address, this will happen soon. The code is as follows:
```
        bool done = false;                        // Event handler flips it to true
 	mg_log_set("4");                          // Set to 0 to disable debug

 	while (true)
 	{
		struct mg_mgr mgr;       
 		done = false;
 		mg_mgr_init(&mgr);                        // Initialise event manager	
		mgr.dns4.url = "udp://127.0.0.1";
 		mg_http_connect(&mgr, s_url, fn, &done);  // Create client connection
		while (!done)
		{
			mg_mgr_poll(&mgr, 1000);    // Infinite event loop
		}
		mg_mgr_free(&mgr);                        // Free resources
 		std::this_thread::sleep_for(std::chrono::milliseconds(1000));
 	}
```
mongoose has helped me a lot and can be easily integrated into my device code. However, after upgrading from version 6.xx to 7.0, , there are some problems with my code, I can't refer to the experience of 6.xx,, so I may have used mongoose inappropriately. Thank you for your reply.

beyondkyq, 2021-01-21:
Maybe I misunderstood it. When the MG_EV_CLOSE event is triggered in dns_cb, mongoose will traverse the s_reqs list to mg_dns_free all the dns_data instead of mg_dns_free  the dns_data corresponding to mg_connection. Does this mean that if there are concurrent DNS resolution operations, there will be problems. mongoose code as follows:
```
if (ev == MG_EV_CLOSE) {
    for (d = s_reqs; d != NULL; d = tmp) {
      tmp = d->next;
      mg_dns_free(d);
    }
  }
```

cpq, 2021-01-21:
This snippet is triggered when a DNS socket gets closed. It happens only when you're freeing the manager, `mg_mgr_free()`. The purpose is to cancel all outstanding requests.

The concurrent DNS operations should not be an issue, because each request has a unique ID.

Thanks to your elaborate report and example, the issue was pinpointed quickly: on `MG_EV_POLL`, a correct requests list was not traversed at all, therefore no timeout checks were performed.

Pushed https://github.com/cesanta/mongoose/commit/6ecb6c150033a2701c25c5ffb64769190b38c485 which fixes this issue in `dns.c` and also adds a unit test to make sure that codepath is triggered.

closed 2021-01-21 by cpq