cesanta/mongoose #1195: Speed Issue

toutant, 2021-01-15:
So mongoose runs very well - and is very solid - easy to use and setup - 
 However there seems to be an issue that is driving me crazier than usual.

I have the latest version of mongoose - but this issue is the same going back to version 6

every second  http request is three times  slower.
First one very fast - second request slow - third request fast again..

Any suggestion ?

cpq, 2021-01-15:
Is there a way to reproduce that?
What do you use - a pre-built binary? Which OS?
Please share your exact environment.

toutant, 2021-01-16:
WIndows 32 on Windows 10 
mongoose MG_VERSION "7.0"
It is highly repeatable - any webpage just load it twice

[image]
[image]

cpq, 2021-01-16:
Do you test it with the console binary (which opens a black console window when server starts),
or a GUI binary (which minimises itself on a system tray) ?

toutant, 2021-01-16:
I took your code as is on github compiled using Visual Studio 2017 it is in a console application as i wanted to test your code.- 
I have tried with prebuilt binary console - same results

I wrote a small pos program which I use xampp for webserver and mysql - which was very fast. no delays at all.

I then tried using your mongoose prebuilt and noticed the speed issue - but thought i would download/compile your code and try it /  find the reason for the speed issue - but no luck.

i have tried small and large files - no difference - i thought it might be buffer related issue or ip lookup issue - but i am not smart enough to figure this out

by the way this is all tested on localhost... 
[image]

cpq, 2021-01-20:
Trying to reproduce that, but have not seen speed issues so far.

Using Windows 10 home edition 64-bit, and 1996 MSVC compiler that produces 32-bit binary.
A simple static website with 3 images:

```sh
-rw-r--r--  1 lsm  staff    80K 20 Jan 16:11 1.png
-rw-r--r--  1 lsm  staff    46K 20 Jan 16:20 2.png
-rw-r--r--  1 lsm  staff    78K 20 Jan 16:26 3.png
-rw-r--r--  1 lsm  staff   216B 20 Jan 16:25 index.html
```

index loads in about 20ms, and images take about 120ms. Tested in a VirtualBox VM run by MacOS, so the VM is pretty slow.
The site is static, whereas you guys report on dynamic content - I see `xhr` mark in the devtools.

So could you please: 
1. Clarify on whether an issue happens on a static content?
2. If it only happens on dynamic requests, please share your code.

[image]

toutant, 2021-01-20:
it is static  page -
i created a test page which loads  html scripts and css (local scripts local css ) - no javascript running on load - no pictures - no google fonts - no php - 
and every second load is longer.

I also tested with merging all of the javascript files and css and build an all inclusive single page to see if that would change the load times - but it doesnt...

toutant, 2021-01-21:
lol

you are not taking this as an issue - i have already moved on to another webserver - 
disappointing - since it is an issue

that was just a sample of one  out of  over 50 test done ... i really thought it was my code ( would not be the first time - been programming for 30 years ) when i realized it was your code then i was initially relieved - now i do not have the time and energy to try and convince people there is an issue.

as explained before - other tests with other web servers DO NOT have this issue - just yours

thank you in advance for what ever amount of time you spent on this issue

cpq, 2021-01-21:
@toutant apologies, did not mean to be offensive at all
Definitely taking this an an issue, just having trouble reproducing it.
If you could share your sample static site, that's be appreciated.

toutant, 2021-01-21:
no worries - like i said the speed issue is deal breaker  for me - I was hoping to use it for POS software but there are some other issues with your software - after a week of running it will freeze up ( on polling section ) no errors reported just locked up

this is so repeatable on my end you should have no trouble identifying the problem

- i have now downloaded and have  been using https://github.com/yhirose/cpp-httplib 
- attached is the network speed report using cpp-httplib ( same file used to test yours )
[image]

cpq, 2021-01-21:
Tried different static sites, wasn't able to trigger an issue you've described. It might be caused by specific sequence of requests made by jquery - again, a sample would be appreciated.

W.r.t. the lockup - that might be due to a bug (that is already fixed) that did not release file descriptors for requests that hit local cache. Could you clarify that you're using the latest codebase?

toutant, 2021-01-22:
- the static page I used is all javascript/css  loading of  custom POS library I do not wish to share at this time.
 - no dynamic loading - completly static
 - using latest code from github

Here is what i did to test mongoose - created a local file index.htm add lots of js includes css includes - load it 10 times using mongoose on local host/local IP - every second one was slower - it is not an empty page or a page with very little in it.

there is no dynamic stuff  ( no jquery ) 
For last test loading a page via 
XMLHttpRequest   too simulate loading script file - but it was local file  loaded by mongoose. 

I have moved on to the other library for now - but i will keep checking back to see how mongoose progresses

cpq, 2021-01-22:
@toutant
I might be mistaken, but on all your screenshot, no exceptions, the accesses are "xhr". These are AJAX, meaning, dynamic. Also the very first screenshot says that originator is "jquery".
It is understood that the website has only static content, but apparently JS is doing some work loading things.

Anyways, so far there was no luck reproducing the issue.

Closing this until more clarity.

closed 2021-01-22 by cpq