cesanta/mongoose #1198: State TLS compatibility

lnksz, 2021-01-22:
Hi There,

from your `test.yaml` it is not stable/deterministic to which versions of mbedtls and openssl are you tested/compatible with.

This info would be great so that one can see dependency-compatibility transparently.

```yml
  linux:
    runs-on: ubuntu-latest  # <== moving targen
    steps:
    - uses: actions/checkout@v2
    - name: Install packages
      run: sudo apt-get install libmbedtls-dev
    - name: make
      run: make test test++ IPV6=0
    - name: openssl
      run: make test SSL=OPENSSL IPV6=0
```
mbedtls has 3 relevant branches:
- 2.x bleeding edge
- 2.16 new LTS
- 2.7.x old LTS although this has an EOL in 2021-February
- upcoming v3...

openssl:
- 1.1.1* although I do not know how much of a difference the sub-versions make
- 1.0.1 EOL : a (non-)support statement about it would be beneficent.

I think a section about TLS compatibility in your README/Doc would be great.

cpq, 2021-01-22:
It is intentional that the test and packages are not versioned, therefore they are a moving target.
The idea is to be compatible with whatever the latest stable release of the TLS package is.

The exact version could be easily looked up at e.g.
   https://packages.ubuntu.com/search?keywords=openssl -> https://packages.ubuntu.com/focal/openssl -> 1.1.1f

Does that make sense?

lnksz, 2021-01-22:
> The idea is to be compatible with whatever the latest stable release of the TLS package is.

*whatever the test executor's distro packaged

> Does that make sense?

I understand and respect your choice, but I think communicating this (and e.g. no LTS support, which popped up lately in the issues because of the v7 hard-cut) as part of a "living at head" manifesto would be helpful for adopters. If not a manifesto, a img.shields.io bedge with the currently tested versions from mbedTLS and openssl would be helpful (and fancy :).

Still if you do not agree feel free to close.

cpq, 2021-01-22:
Thanks for the background, it makes sense.

This project has started in 2004, and the OpenSSL API it used, haven't change much since. I believe, some newer API is protected by the relevant conditionals , e.g. https://github.com/cesanta/mongoose/blob/9eabf3fa8171bc729e84868b9ecc715a790cc6a8/src/tls.c#L242
Therefore, whilst not explicitly tested on quite old versions, it would be safe to say that OpenSSL 0.9 onwards is supported.

As for mbedtls, the API subset that is used, makes it compatible with the 2.x branch onwards - can't remember starting from which version, I think I saw it built on 2.6 IIRC.  There were even couple of declaration hard-coded, https://github.com/cesanta/mongoose/blob/9eabf3fa8171bc729e84868b9ecc715a790cc6a8/src/tls.c#L18-L19 , which avoids to use newer includes to make it compatible with the older versions.
So for mbedtls, it is also safe to assume that 2.x are supported.

The conclusion is - Mongoose wants to be compatible with any TLS lib version, currently stable and any "reasonably old one". 
Testing, however, is done on the latest version only - well, whatever latest is Microsoft chooses to use in their Github Actions, as you've correctly pointed out.

I agree that showing the currently used TLS lib versions as badges might be helpful, although I am not sure how to make it automatic. Adding a hardcoded link is going to rot soon - so if you have any hints, please share.

lnksz, 2021-01-22:
I think something like in the readme/docs would be already beneficial:

> it would be safe to say that OpenSSL 0.9 onwards is supported.
> it is also safe to assume that 2.x are supported.
> Mongoose wants to be compatible with any TLS lib version, currently stable and any "reasonably old one".
Testing, however, is done on the latest version only

When I will have the time to play with the github actions and badges, I will send a MR.

closed 2021-01-22 by lnksz

lnksz, 2021-01-22:
Just for fun:

> The exact version could be easily looked up at e.g.
https://packages.ubuntu.com/search?keywords=openssl -> https://packages.ubuntu.com/focal/openssl -> 1.1.1f

But surprise, in github terms "latest" isn't focal (20.04) but bionic (18.04), based on their [virt-env description](https://github.com/actions/virtual-environments#available-environments) (as of 2021-01-22)
> Ubuntu 18.04 | ubuntu-latest
> -- | --

And on the [ubuntu package repo](https://packages.ubuntu.com/search?keywords=openssl):

> bionic (18.04LTS) (utils): Secure Sockets Layer toolkit - cryptographic utility
> 1.1.1-1ubuntu2.1~18.04.7 [security]: amd64 i386
> 1.1.0g-2ubuntu4 [ports]: arm64 armhf ppc64el s390x

lnksz, 2021-01-22:
shields.io has already arch integration. If there will be e.g. ubuntu package repo support you could use that for reference.

`https://img.shields.io/archlinux/v/community/x86_64/mbedtls` => [image]