cesanta/mongoose #1209: File link in directory listing ist not working (MG_ENABLE_DIRECTORY_LISTING)

rhae, 2021-01-31:
The output of the directory listing shows all files.
When clicking on a link the file cannot be downloaded.
The is a slash missing in the link. 
The diff below resolves it.

```
@@ -962,7 +962,7 @@ static void printdirentry(struct mg_connection *c, struct mg_http_message *hm,
   // mg_escape(file_name, path, sizeof(path));
   // href = mg_url_encode(mg_mk_str(file_name));
   mg_http_printf_chunk(c,
-                       "  <tr><td><a href=\"%.*s%s%s\">%s%s</a></td>"
+                       "  <tr><td><a href=\"%.*s/%s%s\">%s%s</a></td>"
                        "<td>%s</td><td>%s</td></tr>\n",
                        (int) hm->uri.len, hm->uri.ptr, name, slash, name, slash,
                        mod, size);
```

cpq, 2021-01-31:
The directory listing works without this change.
Maybe it does not under certain circumstances. What is your OS, compiler, and how do you test it?

rhae, 2021-01-31:
I work on win10, MSVC2019, x86

This is my webroot
index.html
js/serviceworker.js
js/libs

This is the the content of the js directory

```
<!DOCTYPE html><html lang="en"><head><title>Index of /js</title><meta charset="utf-8">
 <meta name="viewport" content="width=device-width, initial-scale=1">
<style>th,td {text-align: left; padding-right: 1em; font-family: monospace; }</style></head><body><h1>Index of /js</h1><table cellpadding="0"><thead><tr><th>Name</th><th>Modified</th><th>Size</th></tr><tr><td colspan="3"><hr></td></tr></thead><tbody>
<tr><td><a href="/jsserviceworker.js">serviceworker.js</a></td><td>25-Jan-2021 20:58</td><td>2.7k</td></tr>
<tr><td><a href="/jslibs/">libs/</a></td><td>26-Jan-2021 23:16</td><td>[DIR]</td></tr>
</tbody><tfoot><tr><td colspan="3"><hr></td></tr></tfoot></table><address>Mongoose v.7.1</address></body></html>
```

Do you see the <a> links? The slash is missing.

cpq, 2021-01-31:
The issue is that the directory URL must end with `/`. Your URL, apparently, does not.
It looks like you have navigated to that URL manually, instead of following Mongoose root page links.
Anyways, the issue is legit, we'll fix it shortly.

commit 582fde5c (2021-01-31): Fix #1209
closed 2021-01-31 by cpq