cesanta/mongoose #1211: [question] try to serve a web in esp32 with spiffs

ihouses, 2021-02-02:
Hi,
I am using a esp32 with spiffs and  I put a index.html in a folder in a way that I can execute in my code:
FILE *file = fopen("/spiffs/web/index.html", "r");
and the file is open normally.

I am following one of the mongoose examples and I set this:

static const char *s_web_directory = "/spiffs/web\0";

at some point this code is executed when I set in my browner the correct address

struct mg_http_serve_opts opts = {.root_dir = s_web_directory};
 mg_http_serve_dir(c, ev_data, &opts);

Then the answer of mongoose in the browner is: Bad web root [/spiffs/web]

I think that this mongoose  function:

// Try to avoid dirent API
static int mg_is_dir(const char *path) {
#if MG_ARCH == MG_ARCH_FREERTOS
  struct FF_STAT st;
  return (ff_stat(path, &st) == 0) && (st.st_mode & FF_IFDIR);
#else
  struct stat st;
  return (stat(path, &st) == 0) && (st.st_mode & S_IFDIR);
#endif
}

does not like my path.

What I am doing wrong with the path?

ShaunLeMouton, 2021-02-05:
Hi, it seems that the problem is not solved in the new version.
`FILE *fp = fopen("/www/web/index.html", "r");`
Work's as expected.

`static const char *s_web_directory = "/www/web\0";
            mg_http_serve_opts opts;
			opts.root_dir = s_web_directory;
mg_http_serve_dir(c, hm, &opts);`

The Browser return Bad web root [/www/web]

I think mongoose cannot handle the spiffs path of the esp32.

cpq, 2021-02-05:
An example https://github.com/cesanta/mongoose/tree/master/examples/esp32 is tested and works.

ihouses, 2021-02-09:
It works for me too,

ShaunLeMouton, 2021-02-09:
Yes you are right. It works. It was my fault.

ShaunLeMouton, 2021-08-05:
I tried the example above. It's not working for me on an esp32 device. The mg_http_serve_dir throws an error:

```
1970-01-01 00:00:23  I main.c:55:app_main        FOUND hello.txt
1970-01-01 00:00:23  I sock.c:468:mg_listen      1 accepting on http://0.0.0.0:80 (port 80)
1970-01-01 00:00:23  I main.c:63:app_main        Starting Mongoose web server v7.4
1970-01-01 00:00:24    sock.c:393:accept_conn    2 accepted 192.168.179.3:51233

***ERROR*** A stack overflow in task main has been detected.

Backtrace:0x40087433:0x3ffb9520 0x40087add:0x3ffb9540 0x40087d91:0x3ffb9560 0x40088a49:0x3ffb95e0 0x40087e84:0x3ffb9600 0x40087e3a:0x00000000 |<-CORRUPTED
0x40087433: panic_abort at C:/Users/mp/esp/esp-idf/components/esp_system/panic.c:330

0x40087add: esp_system_abort at C:/Users/mp/esp/esp-idf/components/esp_system/system_api.c:106

0x40087d91: vApplicationStackOverflowHook at C:/Users/mp/esp/esp-idf/components/freertos/xtensa/port.c:474

0x40088a49: vTaskSwitchContext at C:/Users/mp/esp/esp-idf/components/freertos/tasks.c:2771

0x40087e84: _frxt_dispatch at C:/Users/mp/esp/esp-idf/components/freertos/xtensa/portasm.S:431

0x40087e3a: _frxt_int_exit at C:/Users/mp/esp/esp-idf/components/freertos/xtensa/portasm.S:231

```

Does someone have an idea what's going wrong?

commit 457d76b0 (2021-02-03): Fix #1211 - adjust ESP32 example to serve FS