cesanta/mongoose #1216: [Examples] Web sockets server to update multiple clients

gbThreepwood, 2021-02-07:
I am new to Mongoose, and would like to use it for a websocket application (running on a ESP32) where I broadcast updates to multiple clients. I am looking at the example websocket_chat in version 6.18, which does what I want, but I am not sure how to implement this in 7.1.

The _mg_next()_ function to iterate over active connections is missing in 7.1. Do I use the *next pointer in the mg_connection struct to somehow iterate over all active connections, and run _mg_ws_send()_ on each connection? Or is there some helper functions, or more elegant solution that I have missed?

A more sophisticated websocket server example for version 7 would be greatly appreciated.

cpq, 2021-02-07:
https://github.com/cesanta/mongoose/blob/master/examples/timers/main.c should give a clue.

gbThreepwood, 2021-02-07:
I missed that example, thanks.

> https://github.com/cesanta/mongoose/blob/master/examples/timers/main.c should give a clue.

closed 2021-03-08 by cpq