cesanta/mongoose #1219: Compiling mongoose library on Keil v5

gFavi93, 2021-02-10:
Hello,
I'm trying to integrate mongoose libraries in a project and I use uVision Keil v 5.10 as framework but I have some problems when I compile my code because the following header files are missing : 
<fcntl.h>
<sys/stat.h>
<sys/types.h>
Is there a possibility to compile these libraries in this environment ?

cpq, 2021-02-10:
Could you describe your environment in more detail, please?
What OS, what is the target platform / device ?

gFavi93, 2021-02-10:
The description of the environment is present in the attached image :
[image]
I'm trying to develop web socket communication on STM32 device

cpq, 2021-05-12:
If still relevant - could you try the process described at https://cesanta.com/docs/#custom-build please?

polluks, 2021-07-20:
Keil doesn't seem to support POSIX headers, see https://github.com/payne92/bare-metal-arm/issues/14#issue-248980859.

cpq, 2021-08-23:
While not on Keil, but we've published a bunch on examples for STM32
https://github.com/mongoose-examples

closed 2021-08-23 by cpq