cesanta/mongoose #1222: Problem with raw byte data

marcelwirtz, 2021-02-26:
If i try to reply raw byte data like a compressed string

`static const char *data = "\x00\x00\x00\x04x\xDA\x0BI-.\x01\x00\x03\xDD\x01\xA1";`

the result is always empty

`HTTP/1.1 200 OK..Content-Length: 0....`

[Example.zip](https://github.com/cesanta/mongoose/files/6049729/Example.zip)

nliviu, 2021-02-26:
As a string, your `data` is a nul string.
Try something like this
```
mg_http_reply(c, 200, "", "%.*s", (int)sizeof(data), data);
```

marcelwirtz, 2021-02-26:
Tried it, but got the same result in the example provided above.

cpq, 2021-02-28:
Any printf-like function stops printing a string if it meets a 0 byte. Therefore, any printf-like function, like `mg_http_reply()`, cannot be used with binary data which might have embedded zeros.
Use `mg_printf()` to print headers, then call `mg_send()`  to send binary data as a body.

closed 2021-02-28 by cpq