cesanta/mongoose #1226: Ws message loss from server side

robinyeeh, 2021-03-04:
I'm trying to send message from using ev poll when client connected, but client side always loss one ws frame. I tried to move the code to client side to send, i works fine. Can anyone help on this? Thanks a lot.

conn init source code:
	struct mg_mgr mgr;                        
	mg_mgr_init(&mgr);                        
	mg_http_listen(&mgr, s_listen_on, fn, NULL); 
	for (;;) mg_mgr_poll(&mgr, 10);          
	mg_mgr_free(&mgr);
event handler source code:	
else if (ev == MG_EV_POLL) {
		if (!g_connected) {
			return;
		}

		char msg[256];
		snprintf(msg, 256, "message from server - %d", g_count);
		printf("Send message to server, msg: %s\n", msg);
		mg_ws_send(c, msg, strlen(msg), WEBSOCKET_OP_TEXT);
//		mg_iobuf_delete(&c->send, c->send.len);
		g_count++;
	}

Client side source code:
	struct mg_mgr mgr;      
	bool done = false;        
	struct mg_connection *c; 
	mg_mgr_init(&mgr);        
	c = mg_ws_connect(&mgr, s_url, fn, &done, NULL);     
	while (c && done == false) mg_mgr_poll(&mgr, 10);  
	mg_mgr_free(&mgr);        

	if (ev == MG_EV_WS_MSG) {
		// Got websocket frame. Received data is wm->data. Echo it back!
		struct mg_ws_message *wm = (struct mg_ws_message *) ev_data;
		char msg[256];
		memcpy(msg, wm->data.ptr, wm->data.len);
		printf("Got message from client: %s\n", msg);
	}

As you can see it always losses one ws frame:
GOT ECHO REPLY: [message from server - 30957]
GOT ECHO REPLY: [message from server - 30959]
GOT ECHO REPLY: [message from server - 30961]
GOT ECHO REPLY: [message from server - 30963]
GOT ECHO REPLY: [message from server - 30965]
GOT ECHO REPLY: [message from server - 30967]
GOT ECHO REPLY: [message from server - 30969]
GOT ECHO REPLY: [message from server - 30971]

cpq, 2021-03-04:
Publish an entire event handler functions for client and server. Clearly mark which is which. Delete unused code. Make sure that you test an actual code. Publish logs, too.

cpq, 2021-03-08:
No response, closing.

closed 2021-03-08 by cpq