cesanta/mongoose #1231: Improvement Request

oamc, 2021-03-08:
Hello.
I think it would be very helpful to have listdir, and mg_http_serve_dir to pass headers out of the box. These headers can be helpful in enforcing policies like HSTS everywhere.

cpq, 2021-03-09:
@oamc could you elaborate please?
Could you sketch a quick example?

oamc, 2021-03-10:
`static void listdir(struct mg_connection *c, struct mg_http_message *hm,
                    char *dir, const char *headers);`

`void mg_http_serve_dir(struct mg_connection *, struct mg_http_message *hm,
                       struct mg_http_serve_opts *, const char *headers);`

Passing headers to these functions and subsequent mg_printf or mg_http_reply thereof can help enforcing HSTS and CSP polices. As it is if I were to use the mg_serve_dir function I cannot add custom headers unless I modify or rewrite it as shown above.

cpq, 2021-03-10:
@oamc does it make sense to just add `const char *extra_headers` to the `struct mg_http_serve_opts` ?

cpq, 2021-03-10:
Pushed https://github.com/cesanta/mongoose/commit/6366c832b0faa1fdf29c45d80a4d95054d34d3df
Thank you

closed 2021-03-10 by cpq

oamc, 2021-03-10:
> @oamc does it make sense to just add `const char *extra_headers` to the `struct mg_http_serve_opts` ?

Actually that really makes more sense.