cesanta/mongoose #1235: How to deal with file uploads from normal form request

marcelwirtz, 2021-03-10:
Hey,

i've seen the file upload example and tested a bit with this, but as we have no multipart file uploads anymore, how can we deal with file uploads from a normal form request.

We use libcurl on client side but because we need more information on the request we are using the cUrl Mime post interface, explained here ... https://curl.se/libcurl/c/CURLOPT_MIMEPOST.html

But this results in one file which contains all the parts.

Is there any other way than analyzing the entire file and splitting it up manually?

Kind regards,
Marcel

cpq, 2021-03-10:
I guess the best way is to add a function `mg_http_next_multipart()`, with a signature like this:

```c
int mg_http_next_multipart(struct mg_str body, int offset, struct mg_http_message *part);
```

This function would return the next offset, and populate `part` for the current iteration (headers and part body).
Feel free to put together such function - or wait until we do it.

cpq, 2021-03-17:
Here we go, https://github.com/cesanta/mongoose/tree/master/examples/form-upload

closed 2021-03-17 by cpq

SunnyWeiLucky, 2021-03-18:
Hi, I write code in [testmongoose](https://github.com/SunnyWeiLucky/testmongoose) @cpq