cesanta/mongoose #1237: mongoose.exe configuration + security

Hafplo, 2021-03-11:
Hello,
Experimenting with using pre-built mongoose.exe.
It is super simple and easy to start! Great app.
But have the following issues:
1. it defaults to auto-enabling every time it launches. Is there a way to remove it?
2. it defaults to system root folder. Can we secure it so it default to specific folder + have limited access?
3. it has auto-generated a mongoose.conf file. Is there documentation of the flags?

cpq, 2021-03-11:
1. Let us take a look
2. It looks like that happens when it is auto-started - for some reason, the current working dir in that case is system root. If you start it manually from whatever folder, it serves that folder. Let us take a look at auto-start scenario.
3. Linux or Mac version could list command flags since they are terminal apps:

```
$ ./mongoose -h yes
Mongoose v.7.2.3, built Mar  7 2021 11:44:10
Usage: ./mongoose OPTIONS
  -H yes|no - enable traffic hexdump, default: 'no'
  -S GLOB   - glob pattern for SSI files, default: '#.html'
  -a FILE   - log file, default: ''
  -d DIR    - directory to serve, default: '.'
  -l ADDR   - listening address, default: 'http://0.0.0.0:8000'
  -u PASS   - user password, default: '(null)'
  -v LEVEL  - debug level, from 0 to 4, default: '2'
```

cpq, 2021-03-20:
Pushed version 7.3.1, where issues (1) and (2) have been addressed.

Now, the autorun is enabled automatically when mongoose.exe is first started in a given directory.
If user explicitly disables autorun, the "read-only" flag is set on the executable, which is a marker to prevent enabling autorun at start - and autorun continues to be disabled, unless user changes that.

The autorun current directory is system folder, therefore Mongoose now changes its current dir to the dir where executable lives, which solves issue (2).

closed 2021-03-20 by cpq