cesanta/mongoose #1239: setsockopts introduced in v7

panosto, 2021-03-11:
Hi

I have a question and a possible request regarding `setsockopts()` function which was introduced in v7.

Why is this function called in the `mg_listen()` as well?
In my custom architecture (bsd tcpip), the stack is dropping the listener if there is no incoming connection at all after startup and the keepalive timer times out.  

Can you consider of making these options configurable through the api in the next revision?
If not, I think it makes sense to provide a build option to disable them completely. Or at least disable them when `MG_ARCH_CUSTOM ` is used. 

Differences like this between v6 and v7, are not just refactoring, but change the behavior, which make the upgrade in our applications more difficult or can reveal surprises in the field which are not expected.

Thank you for your consideration
Panagiotis

cpq, 2021-03-13:
@panosto you're correct, apparently for listening sockets, setsockopts() should not be called

cpq, 2021-03-13:
Pushed https://github.com/cesanta/mongoose/commit/381d7cce5428f48800d82cc2106073b892d061ee

closed 2021-03-13 by cpq