cesanta/mongoose #1241: How to upload file to server using http client

SunnyWeiLucky, 2021-03-12:
Hi,I read many exampes with mongoose.But I can not find how to upload file to server using http client.Can you help me,give me a example?

cpq, 2021-03-13:
@SunnyWeiLucky do you mean, upload a file using mongoose, do some external web server?

SunnyWeiLucky, 2021-03-15:
Yes, I want to upload a file using mongoose to other web Server。Do the  moongoose  has the function ?

cpq, 2021-03-15:
See https://github.com/cesanta/mongoose/blob/master/examples/http-client/main.c#L36

closed 2021-03-15 by cpq

SunnyWeiLucky, 2021-03-18:
Hi, @cpq  I test the code using the http-client/main.c example,the result is not ideal。I test using http client and http server ,the are both created by mongose.In the server I write data using the function  mg_http_upload that wirtes **http body data**.When I upload file whose type is like txt, json eg text.I can deal with the boundary in body data by myself.But when I upload file whose type is like pdf,mp4 .I deal with the boundary in body data in same methond failed.I don't why, Can you give me a idea?

cpq, 2021-03-18:
Please share your code

SunnyWeiLucky, 2021-03-18:
Hi, I write code in [testmongoose](https://github.com/SunnyWeiLucky/testmongoose) @cpq .You can see network dir

cpq, 2021-03-18:
The multipart example code removed from the example.
It is discouraged to hand-writing multipart upload, especially if the receiving side is mongoose.

If you want to handle multipart on the server side, please see an example at https://github.com/cesanta/mongoose/tree/master/examples/form-upload

zhoujinhai, 2021-03-24:
> Yes, I want to upload a file using mongoose to other web Server。Do the moongoose has the function ? @cpq

I also have the question(mongoose as a client and upload a file to sever). Do you have solved it?

SunnyWeiLucky, 2021-03-24:
You can see the example [testmongoose](https://github.com/SunnyWeiLucky/testmongoose) .The demo is in network dir.It is a simple example to upload file .I using mongoose as client and server. @zhoujinhai

zhoujinhai, 2021-03-24:
> The multipart example code removed from the example.
> It is discouraged to hand-writing multipart upload, especially if the receiving side is mongoose.
> 
> If you want to handle multipart on the server side, please see an example at https://github.com/cesanta/mongoose/tree/master/examples/form-upload

Hi, @cpq , I want to upload a file to other sever,  and have try follow code, but it not work!
```c++
const char* model_path = "E:/code/python_web/MeshCNN/test_models/test.obj";
 size_t file_size = mg_file_size(model_path);
 char *file_data = mg_file_read(model_path);
 const char* boundary = "----BoundaryNetworkf912decMA4YWxkTrZu0gW";
 char tmp[256];
 snprintf(tmp, sizeof(tmp), 
     "%s\r\n"
     "Content-Disposition: form-data; name=file; filename=%s\r\n"
     "Content-Type: application/octet-stream"
     "\r\n",
     boundary, model_path);
 printf("file_size: %u\n", file_size);

 mg_printf(c,
     "POST %s HTTP/1.0\r\n"
     "Host: %.*s\r\n"
     "Content-Length: %lu\r\n"
     "Content-Type: multipart/form-data; boundary=%s\r\n"
     "\r\n",
     mg_url_uri(s_url), (int)host.len, host.ptr,
     (unsigned long)(file_size + strlen(tmp) + strlen(boundary) + 2),
     boundary);
 mg_printf(c, "%s", tmp);
 mg_send(c, file_data, file_size);
 mg_printf(c, "%s\r\n", boundary);
 free(file_data);
```

The sever output is below and raise a error: `Could not find starting boundary`
```bash
headers:  <CIMultiDictProxy('Host': '192.168.102.116', 'Content-Length': '364917', 'Content-Type': 'multipart/form-data; boundary=----BoundaryNetworkf912decMA4YWxkTrZu0gW')>
Could not find starting boundary b'------BoundaryNetworkf912decMA4YWxkTrZu0gW'
```

I am use python(`aiohttp`) is ok, Do you have any suggestions for clients to upload files?
```python
        async with aiohttp.ClientSession() as session:
            async with session.post(
                    url='http://{}:{}/recognize'.format(web_ip, web_port),
                    data=file_data,
                    # headers={'Content-Type': 'multipart/form-data; boundary=--dd7db5e4c3bd4d5187bb978aef4d85b1'}
            ) as resp:
```

zhoujinhai, 2021-03-24:
> You can see the example [testmongoose](https://github.com/SunnyWeiLucky/testmongoose) .The demo is in network dir.It is a simple example to upload file .I using mongoose as client and server. @zhoujinhai

Yes,  I've read your code，but I have an error when I upload file:
```bash
headers:  <CIMultiDictProxy('Host': '192.168.102.116', 'Content-Length': '364917', 'Content-Type': 'multipart/form-data; boundary=----BoundaryNetworkf912decMA4YWxkTrZu0gW')>
Could not find starting boundary b'------BoundaryNetworkf912decMA4YWxkTrZu0gW'
```

SunnyWeiLucky, 2021-03-24:
The char array of tmp has a error.The http header of "Content-Type: application/octet-stream" should using "Content-Type: application/octet-stream **\r\n**".You can try again

zhoujinhai, 2021-03-24:
> The char array of tmp has a error.The http header of "Content-Type: application/octet-stream" should using "Content-Type: application/octet-stream **\r\n**".You can try again

Yes, I have try and get the same error, 
```C++
snprintf(tmp, sizeof(tmp), 
             "%s\r\n"
             "Content-Disposition: form-data; name=file; filename=%s\r\n"
             "Content-Type: application/octet-stream \r\n"
             "\r\n",
             boundary, model_path);
```