cesanta/mongoose #1244: error in use 'mg_ws_upgrade'

siam7tong, 2021-03-17:
Sorry to bother you, but i have some questions when using 'mg_ws_upgrade'.
I push my video frame to web through websocket, and the main code is as below:

part 1 (in http ev_handler):
` mg_ws_upgrade(c, hm, NULL);`

part 2 (send frame in loop thread every 10 ms):
` mg_ws_send(c, data, sess->wsbuf_len[(sess->rd) % 2], WEBSOCKET_OP_BINARY);`

part 3 (poll every 1 ms):
`mg_mgr_poll(&mgr, 1);`

I print the conn-send-buf-len before one video frame is sent.
line 4284 in mongoose.c, add
`  printf("before send,len=%d\n", c->send.len);`

and terminal output:
>before send,len=0
before send,len=0
before send,len=0
before send,len=0
before send,len=0
before send,len=0

but quickly become:
>*** Error in `./webs': double free or corruption (!prev): 0xb4a03a60 ***

total process shutdown;
or
>2021-03-17 07:36:32  E mongoose.c:2739:ll_write  9 tuc -1/80854 14`  

send process stops and send-buf grows.

I find that errno 14 means ' Bad address', but I don't know where is wrong.
Do you have any suggestion?
Thanks for your kind attention and look forward your prompt reply.

cpq, 2021-03-17:
Please publish your code
Remove all irrelevant parts, leave only what matters: all functions where `mg_*` API is called.

siam7tong, 2021-03-17:
Thanks for your reply.
 My code is here, and the entrance is at the end 'http_task'

```
static int flv_ws_send(void *p) 
{
    flv_ws_params_t *param = (flv_ws_params_t *) p;
    struct mg_connection *c = (struct mg_connection *) param->c;
    ...

    if (...) {
        ...
        bytes = mg_ws_send(c, data, sess->wsbuf_len[(sess->rd) % 2], WEBSOCKET_OP_BINARY);
    }
    ...
    return bytes;
}

static int flv_send_header(void *p) { ... return flv_ws_send(p); }

static int flv_send_tag(void *p) { ... return flv_ws_send(p); }

static void *flv_thread(void *p) {
    ...
    while (1) {

        if (...)  flv_send_header(p);
           
        flv_send_tag();
        
        usleep(10 * 1000);
    }

    return NULL;
}

int flv_ws(struct mg_connection *c, struct mg_http_message *hm) {

    ...
    pthread_create(&sess->thread_id, NULL, flv_thread, params);

    return 0;
}

...

static void ev_handler(struct mg_connection *c, int ev, void *ev_data, void *fn_data) {
    ...
    switch (ev) {
        case MG_EV_HTTP_MSG:{
            struct mg_http_message *hm = (struct mg_http_message *) ev_data;
            ...
            if (mg_http_match_uri(hm, "/api/flv#")) {
                mg_ws_upgrade(c, hm, NULL);
                flv_ws(c, hm);
            }
                ...

            break;
        }
        ...
    }
}

static void *http_task(void *parm) {
    ...
    struct mg_mgr mgr;
    struct mg_connection *c;
    const char *err = NULL;

    mg_mgr_init(&mgr);

    c = mg_http_listen(&mgr, url, ev_handler, &mgr);

    while (c) {
        mg_mgr_poll(&mgr, 1);
    }

    mg_mgr_free(&mgr);
    
    return NULL;
}
```

cpq, 2021-03-17:
Please read the docs: https://cesanta.com/docs/
```
Since the Mongoose's core is not protected against concurrent accesses, make sure that all mg_* API functions are called from the same thread or RTOS task.
```

If you want to offload requests to different threads, use method from https://github.com/cesanta/mongoose/tree/master/examples/multi-threaded

closed 2021-03-17 by cpq