cesanta/mongoose #1246: Multipart File Upload with files > 3mb

marcelwirtz, 2021-03-18:
Hey,

i tested the form upload example with the new multipart function today (thanks for adding this) but i get the following behavior with files larger than 3 mb.

Client side:
```
curl -F "userid=1" -F "caseid=10" -F "file=@S:\\ISO-IMAGES\\DiskDirectorSuite10.iso" localhost:8000/upload
  % Total    % Received % Xferd  Average Speed   Time    Time     Time  Current
                                 Dload  Upload   Total   Spent    Left  Speed
 46 15.7M    0     0   46 7424k      0  1420k  0:00:11  0:00:05  0:00:06 1421k
curl: (56) Recv failure: Connection was reset
```

Server side:
```
2021-03-18 11:34:29    mongoose.c:3065:accept_co 6 accepted 127.0.0.1:57654
2021-03-18 11:34:34    mongoose.c:2805:ll_read   6 tuc 0/0 0 1
2021-03-18 11:34:34  I main.c:12:cb              New request to: [/upload]
2021-03-18 11:34:34  I main.c:20:cb              Name: [userid] Filename: [] Body: [1]
2021-03-18 11:34:34  I main.c:20:cb              Name: [caseid] Filename: [] Body: [10]
2021-03-18 11:34:34  I main.c:20:cb              Name: [file] Filename: [DiskDirectorSuite10.iso] Body: []
2021-03-18 11:34:34    mongoose.c:2945:close_con 6 closed
```

As you can see the ISO image i used here is around 16 mb but it is reproducible with file around 3,5 mb. With files smaller than 3 mb it seems to work
```
curl -F "userid=1" -F "caseid=10" -F "file=@S:\\procexp.exe" localhost:8000/upload
  % Total    % Received % Xferd  Average Speed   Time    Time     Time  Current
                                 Dload  Upload   Total   Spent    Left  Speed
100 2692k  100    10  100 2692k      1   357k  0:00:10  0:00:07  0:00:03  550kThank you!
```

```
2021-03-18 11:37:23    mongoose.c:3065:accept_co 7 accepted 127.0.0.1:57729
2021-03-18 11:37:31  I main.c:12:cb              New request to: [/upload]
2021-03-18 11:37:31  I main.c:20:cb              Name: [userid] Filename: [] Body: [1]
2021-03-18 11:37:31  I main.c:20:cb              Name: [caseid] Filename: [] Body: [10]
2021-03-18 11:37:31  I main.c:20:cb              Name: [file] Filename: [procexp.exe] Body: [MZÉ]
2021-03-18 11:37:31    mongoose.c:2805:ll_read   7 tuc 0/2757632 0 1
2021-03-18 11:37:31    mongoose.c:2945:close_con 7 closed
```

Thanks for further investigation. I will try to track this down too.

Kind regards,
Marcel Wirtz

cpq, 2021-03-18:
Added more comments

https://github.com/cesanta/mongoose/blob/master/examples/form-upload/main.c#L10

marcelwirtz, 2021-03-19:
Okay seems to work, thanks. Is the high CPU usage while uploading normal and what is causing this?

cpq, 2021-03-19:
No idea about high CPU. It might be normal if both sides are on localhost, and the whole thing becomes CPU bound.

closed 2021-03-19 by cpq