cesanta/mongoose #1248: Missing features in v7 compared to v6

jcorporation, 2021-03-19:
First: I like the new api of mongoose 7.x, but I am missing some features form the 6.x series:

1. defining extra mime types
2. webdav server support
3. enable and disable directory listing per directory
4. ip acl
5. rewrite support - I saw #1184 - but implementing it manually feels like a regression

Is there any plan to add some of this features? Or is there a list what features are dropped?

cpq, 2021-03-19:
Out of those, the "extra mime types" sounds  plausible , the rest won't be added.

BTW, I don't think (3) is 6.x. There is a possibility to hide certain files by pattern. But not enable/disable listing.

closed 2021-03-19 by cpq

jcorporation, 2021-03-20:
> BTW, I don't think (3) is 6.x. There is a possibility to hide certain files by pattern. But not enable/disable listing.

I have used the `enable_directory_listing` option in the `mg_serve_http_opts` struct to implement this. I serve multiple document roots through different paths.

cpq, 2021-03-22:
@jcorporation right, correct.

So, those two knobs, `extra_mime_types` and `enable_directory_listing`, are going to be in `struct mg_http_serve_opts`.
The rest of the mentioned features, like WebDav, are gone for good.