cesanta/mongoose #1249: libasan error in line 18 of iobuf.c

jcorporation, 2021-03-20:
I compile my project with integrated mongoose with libasan to detect memory errors. Since mongoose 7 I get the following error for the first http request:

`src/iobuf.c:18:7: runtime error: null pointer passed as argument 2, which is declared to never be null`

I can reproduce this error with the example http-server.

Compile time options added to your example: `-fsanitize=address \
    -fsanitize=undefined -fsanitize=shift -fsanitize=integer-divide-by-zero -fsanitize=unreachable -fsanitize=vla-bound \
    -fsanitize=null -fsanitize=return -fsanitize=signed-integer-overflow -fsanitize=bounds -fsanitize=bounds-strict \
    -fsanitize=alignment -fsanitize=object-size -fsanitize=float-divide-by-zero -fsanitize=float-cast-overflow \
    -fsanitize=nonnull-attribute -fsanitize=returns-nonnull-attribute -fsanitize=bool -fsanitize=enum -fsanitize=vptr -static-libasan`

cpq, 2021-03-20:
@jcorporation thanks. Could you also show iobuf.c snippet please? Line 18 could be different in different versions.

jcorporation, 2021-03-20:
Version is current master, snippet: `memcpy(p, io->buf, io->size < new_size ? io->size : new_size);`

cpq, 2021-03-21:
@jcorporation Current master has `free()` call on line 18

[image]

cpq, 2021-03-21:
Works OK on Mac:

```sh
$ make -C examples/http-server/
cc ../../mongoose.c main.c -I../.. -I../.. -W -Wall -DMG_ENABLE_IPV6=1 -DMG_ENABLE_LINES=1 -DMG_ENABLE_DIRECTORY_LISTING=1 -DMG_ENABLE_SSI=1 -O2 -fsanitize=address -fsanitize=undefined -fsanitize=shift -fsanitize=integer-divide-by-zero -fsanitize=unreachable -fsanitize=vla-bound -fsanitize=null -fsanitize=return -fsanitize=signed-integer-overflow -fsanitize=bounds -fsanitize=alignment -fsanitize=object-size -fsanitize=float-divide-by-zero -fsanitize=float-cast-overflow -fsanitize=nonnull-attribute -fsanitize=returns-nonnull-attribute -fsanitize=bool -fsanitize=enum -fsanitize=vptr -static-libsan -o example
./example
2021-03-21 10:48:59  I sock.c:484:mg_listen      1 accepting on http://localhost:8000
2021-03-21 10:48:59  I main.c:67:main            Starting Mongoose v7.3, serving [.]
```

jcorporation, 2021-03-21:
I freshly cloned your repository and took aboth makefile. There is a typo in the makefile: `-static-libsan` must be `-static-libasan`.

I can reproduce the error:
`src/iobuf.c:18:7: runtime error: null pointer passed as argument 2, which is declared to never be null`

Sorry, I reported the false snippet. Line 18 with `free(io->buf);` is the problematic one.

Plattform:
- Ubuntu 20.10
- x86_64
- gcc 10.2.0
- libasan6 - 10.2.0-13ubuntu1

cpq, 2021-03-22:
Thank you.

Could you confirm that you see the issue with the following:

```
$ make linux
```

Please use the following Makefile:

```make
PROG ?= example
ROOT ?= $(realpath $(CURDIR)/../..)
DEFS ?= -DMG_ENABLE_LINES=1 -DMG_ENABLE_DIRECTORY_LISTING=1 -DMG_ENABLE_SSI=1
CFLAGS ?= -I../.. -W -Wall -DMG_ENABLE_IPV6=1 $(DEFS) $(EXTRA)
VCFLAGS = /nologo /W3 /O2 /I../.. $(DEFS) $(EXTRA) /link /incremental:no /machine:IX86
VC98 = docker run -it --rm -e Tmp=. -v $(ROOT):$(ROOT) -w $(CURDIR) mdashnet/vc98
LIN = docker run -it --rm -v $(ROOT):$(ROOT) -w $(CURDIR) mdashnet/cc2

all: $(PROG)
	$(DEBUGGER) ./$(PROG) $(ARGS)

$(PROG): main.c Makefile
	$(CC) ../../mongoose.c main.c -I../.. $(CFLAGS) -o $@

mongoose.exe: main.c
	$(VC98) wine cl ../../mongoose.c main.c $(VCFLAGS) ws2_32.lib /out:$@

linux: all
linux: CFLAGS += -O2 -fsanitize=address,undefined,shift,null,return,bounds,alignment,object-size,bool,enum -static-libasan
#linux: CC = $(LIN) cc
#linux: DEBUGGER = $(LIN)

clean:
	rm -rf $(PROG) *.o *.dSYM *.gcov *.gcno *.gcda *.obj *.exe *.ilk *.pdb mongoose mongoose_* mongoose.*
```

If an issue reproduces, you could uncomment those lines:
```
#linux: CC = $(LIN) cc
#linux: DEBUGGER = $(LIN)
```

Those are going to build under linux, Ubuntu 20.04, gcc 9.3 - and I see no issues in that environment. Maybe only latest asan library catches an issue in iobuf. Let's find out.

cpq, 2021-03-22:
Also, please change iobuf.c and see if the issue goes away:

```c
  memcpy(p, io->buf, io->size < new_size ? io->size : new_size);
```
to
```c
      size_t len = new_size < io->len ? new_size : io->len;
      if (len > 0) memcpy(p, io->buf, len);
```

jcorporation, 2021-03-22:
commit ab1c924388247c8eff081381645440a6b378b003 fixes the error. Thanks for fixing this issue so fast.

closed 2021-03-22 by jcorporation