cesanta/mongoose #1251: Multipart uploads high CPU usage

marcelwirtz, 2021-03-22:
Hey,

i mentioned in the other issue requests that there is high CPU usage while transfer a file with multipart uploading. This also happens if server an client are not on the same host.

Next thing is if i upload a larger file (300 mb) the more i upload the slower the upload gets which results that i can upload 56.3 MB in 33 minutes and calculated duration counts up and up, while the upload speed is at 0 and only sends a few kilobytes every few seconds.

I have set mongoose to 

`MG_MAX_RECV_BUF_SIZE=(700*1024*1024)`

with the multipart upload example

and tested the upload via curl.exe with the following parameters

`curl -F 'file=@S:/debian-10.4.0-amd64-netinst.iso' http://192.168.253.41:8000`

The ISO has 336M in total.

Hope you are able to reproduce that.

Kind regards,
Marcel

cpq, 2021-03-22:
What if you build with `-DMG_IO_SIZE=524288` ?

Build options are listed at https://cesanta.com/docs/#build-options

marcelwirtz, 2021-03-23:
This seems to work, the CPU usage came down a bit and the memory consumption grows with the amount of transferred data. We limited the bandwidth too to around 1mbit which results in CPU usage around 1-2% while transferring the file.

closed 2021-03-23 by marcelwirtz