cesanta/mongoose #1252: libasan reports signed integer overflow

jcorporation, 2021-03-23:
I freshly cloned current master and tried the example http-server on an arm 32 bit plattform (raspberry 4 with dietpi - armv7l):

```
# make linux
cc ../../mongoose.c main.c -I../.. -I../.. -W -Wall -DMG_ENABLE_IPV6=1 -DMG_ENABLE_LINES=1 -DMG_ENABLE_DIRECTORY_LISTING=1 -DMG_ENABLE_SSI=1  -O2 -fsanitize=address,undefined,shift,null,return,bounds,alignment,object-size,bool,enum -static-libasan -o example
./example 
2021-03-23 07:36:55  I sock.c:484:mg_listen      1 accepting on http://localhost:8000
2021-03-23 07:36:55  I main.c:78:main            Starting Mongoose v7.3, serving [.]
src/util.c:336:20: runtime error: signed integer overflow: 1616485016 * 1000 cannot be represented in type 'long int'
```

Problematic line should be: `return ts.tv_sec * 1000 + ts.tv_nsec / 1000000;`

cpq, 2021-03-24:
Thank you.

Could you change the problematic line to the following and rebuild:

```c
  return (unsigned long) ((uint64_t) ts.tv_sec * 1000 + ts.tv_nsec / 1000000);
```

jcorporation, 2021-03-24:
This change fixes the issue.

closed 2021-03-24 by jcorporation