cesanta/mongoose #1253: libasan reports global-buffer-overflow for http-client example if uri contains //

jcorporation, 2021-03-23:
Change uri in htttp-client example to: `http://info.cern.ch//te`

Compile it with libasan (same parameters as for http-server linux target):
```
make linux
cc ../../mongoose.c -I../.. -W -Wall -DMG_ENABLE_LINES  -O2 -fsanitize=address,undefined,shift,null,return,bounds,alignment,object-size,bool,enum -static-libasan -o example main.c
./example 
2021-03-23 19:17:36    sock.c:379:mg_connect     1 -> http://info.cern.ch//te
2021-03-23 19:17:36    sock.c:379:mg_connect     2 -> udp://8.8.8.8:53
=================================================================
==7833==ERROR: AddressSanitizer: global-buffer-overflow on address 0x5646dae0e998 at pc 0x5646dadd88bd bp 0x7fffb0b95d30 sp 0x7fffb0b95d20
READ of size 1 at 0x5646dae0e998 thread T0
    #0 0x5646dadd88bc in mg_dns_send (/home/juergen/projekte/myMPD/mongoose/examples/http-client/example+0x12e8bc)
    #1 0x5646dade22ba in mg_sendnsreq.constprop.0 (/home/juergen/projekte/myMPD/mongoose/examples/http-client/example+0x1382ba)
    #2 0x5646dade3132 in mg_connect (/home/juergen/projekte/myMPD/mongoose/examples/http-client/example+0x139132)
    #3 0x5646dade34ed in mg_http_connect (/home/juergen/projekte/myMPD/mongoose/examples/http-client/example+0x1394ed)
    #4 0x5646dacec4e6 in main (/home/juergen/projekte/myMPD/mongoose/examples/http-client/example+0x424e6)
    #5 0x7fb951f9fcb1 in __libc_start_main (/lib/x86_64-linux-gnu/libc.so.6+0x28cb1)
    #6 0x5646dacec67d in _start (/home/juergen/projekte/myMPD/mongoose/examples/http-client/example+0x4267d)

0x5646dae0e998 is located 0 bytes to the right of global variable '*.LC9' defined in 'main.c' (0x5646dae0e980) of size 24
  '*.LC9' is ascii string 'http://info.cern.ch//te'
SUMMARY: AddressSanitizer: global-buffer-overflow (/home/juergen/projekte/myMPD/mongoose/examples/http-client/example+0x12e8bc) in mg_dns_send
Shadow bytes around the buggy address:
  0x0ac95b5b9ce0: 00 00 00 00 00 00 00 00 00 00 00 00 00 00 00 00
  0x0ac95b5b9cf0: 00 00 00 00 00 00 00 00 00 00 00 00 00 00 00 00
  0x0ac95b5b9d00: 00 00 00 00 00 00 00 00 07 f9 f9 f9 f9 f9 f9 f9
  0x0ac95b5b9d10: 00 00 00 00 f9 f9 f9 f9 05 f9 f9 f9 f9 f9 f9 f9
  0x0ac95b5b9d20: 02 f9 f9 f9 f9 f9 f9 f9 07 f9 f9 f9 f9 f9 f9 f9
=>0x0ac95b5b9d30: 00 00 00[f9]f9 f9 f9 f9 00 00 00 00 00 00 00 00
  0x0ac95b5b9d40: 00 00 00 00 00 00 00 00 00 00 00 00 00 00 00 00
  0x0ac95b5b9d50: 00 00 00 00 00 00 00 00 00 00 00 00 00 00 00 00
  0x0ac95b5b9d60: 00 00 00 00 00 00 00 00 00 00 00 00 00 00 00 00
  0x0ac95b5b9d70: 00 00 00 00 00 00 00 00 00 00 00 00 00 00 00 00
  0x0ac95b5b9d80: 00 00 00 00 00 00 00 00 00 00 00 00 00 00 00 00
Shadow byte legend (one shadow byte represents 8 application bytes):
  Addressable:           00
  Partially addressable: 01 02 03 04 05 06 07 
  Heap left redzone:       fa
  Freed heap region:       fd
  Stack left redzone:      f1
  Stack mid redzone:       f2
  Stack right redzone:     f3
  Stack after return:      f5
  Stack use after scope:   f8
  Global redzone:          f9
  Global init order:       f6
  Poisoned by user:        f7
  Container overflow:      fc
  Array cookie:            ac
  Intra object redzone:    bb
  ASan internal:           fe
  Left alloca redzone:     ca
  Right alloca redzone:    cb
  Shadow gap:              cc
==7833==ABORTING

```

cpq, 2021-03-24:
It was an issue in `mg_url_uri()`.
Please fetch the latest version and retry.

jcorporation, 2021-03-24:
Issue is fixed with latest version. Many thanks for your fast response!

closed 2021-03-24 by jcorporation