cesanta/mongoose #1254: Segmentation fault in mg_http_reply

jcorporation, 2021-03-25:
This issue was detect by my api fuzzer for my myMPD project that integrates mongoose.

The %n format specifiers seems to be the trigger of the segmentation fault.

Steps to reproduce:
- change your http-server example to reply with: `mg_http_reply(c, 200, "Content-Type: application/json\r\n", "%n%n");` instead of serving the directory
- open the site in the browser

Plattform:
- x86_64
- Ubuntu 20.10

Error:
```
$ make linux
./example 
2021-03-25 20:59:07  I sock.c:484:mg_listen      1 accepting on http://localhost:8000
2021-03-25 20:59:07  I main.c:80:main            Starting Mongoose v7.3, serving [.]
AddressSanitizer:DEADLYSIGNAL
=================================================================
==22392==ERROR: AddressSanitizer: SEGV on unknown address 0x000000000011 (pc 0x7f2064ae7380 bp 0x7fff003c6230 sp 0x7fff003c5ca0 T0)
==22392==The signal is caused by a WRITE memory access.
==22392==Hint: address points to the zero page.
    #0 0x7f2064ae7380  (/lib/x86_64-linux-gnu/libc.so.6+0x76380)
    #1 0x7f2064af9879  (/lib/x86_64-linux-gnu/libc.so.6+0x88879)
    #2 0x558cb8b4972d in __vsnprintf_chk (/home/juergen/projekte/myMPD/mongoose/examples/http-server/example+0x8172d)
    #3 0x558cb8bf781b in mg_vasprintf (/home/juergen/projekte/myMPD/mongoose/examples/http-server/example+0x12f81b)
    #4 0x558cb8c08fab in mg_http_reply (/home/juergen/projekte/myMPD/mongoose/examples/http-server/example+0x140fab)
    #5 0x558cb8c0edad in http_cb (/home/juergen/projekte/myMPD/mongoose/examples/http-server/example+0x146dad)
    #6 0x558cb8be3d7c in mg_call (/home/juergen/projekte/myMPD/mongoose/examples/http-server/example+0x11bd7c)
    #7 0x558cb8c03606 in mg_mgr_poll (/home/juergen/projekte/myMPD/mongoose/examples/http-server/example+0x13b606)
    #8 0x558cb8b0ebcc in main (/home/juergen/projekte/myMPD/mongoose/examples/http-server/example+0x46bcc)
    #9 0x7f2064a99cb1 in __libc_start_main (/lib/x86_64-linux-gnu/libc.so.6+0x28cb1)
    #10 0x558cb8b0ef9d in _start (/home/juergen/projekte/myMPD/mongoose/examples/http-server/example+0x46f9d)

AddressSanitizer can not provide additional info.
SUMMARY: AddressSanitizer: SEGV (/lib/x86_64-linux-gnu/libc.so.6+0x76380) 
==22392==ABORTING
make: *** [Makefile:10: all] Error 1

```

gegzy, 2021-03-25:
You are missing arguments to the function call. In your example, you are passing in "%n%n", but not passing in any further variables to cater for that. More importantly, %n requires a pointer to a signed int.

e.g.
```
int var1, var2;
mg_http_reply(c, 200, "Content-Type: application/json\r\n", "%n%n", &var1, &var2);
```

jcorporation, 2021-03-26:
mg_http_reply treats this argument as an format string? Is this undocumented?

cpq, 2021-03-26:
@jcorporation That is documented, https://cesanta.com/docs/#mg_http_reply

jcorporation, 2021-03-26:
Sorry, my error

closed 2021-03-26 by jcorporation