cesanta/mongoose #1255: Github CodeQL alert for mg_http_write_chunk

jcorporation, 2021-03-29:
Following code produces an CodeQL alert:
```
void mg_http_write_chunk(struct mg_connection *c, const char *buf, size_t len) {
  mg_printf(c, "%X\r\n", len);
  mg_send(c, buf, len);
  mg_send(c, "\r\n", 2);
}
```

len: This argument should be of type 'unsigned int' but is of type 'unsigned long'

`mg_printf(c, "%X\r\n", len);` I think this should be `mg_printf(c, "%lu\r\n", len);`

lnksz, 2021-03-29:
`size_t` has different width on different platforms..., if decimal output is an option `%zu` is dedicated for size_t, but to keep the original formatting `%zX` would do the trick. (the %z is a C99 feature)

cpq, 2021-04-29:
Some stdlibs do not like `%z` , so casting to `unsigned long` to squash the warning..

commit 75857287 (2021-04-29): Fix #1255
closed 2021-04-29 by cpq