cesanta/mongoose #1257: mg_socketpair doesn't have the same functionality as mg_broadcast

mcottrell1, 2021-04-01:
With mongoose 6.x I was handling http requests in a thread and using mg_broadcast to wake up the poll thread when the response was ready to be sent. This functionality appears to be gone in 7.x as mg_socketpair doesn't appear to offer any way to wake up an ongoing select call. I have looked at the source and the multi threaded example cited in other tickets related to mg_broadcast.

Is there any plan to bring this functionality back? Or is it still present and I'm missing something? I would rather not be required to use a small select timeout and rely on it timing out in order to be able to send a response in a timely manner.

cpq, 2021-04-02:
Please try to implement your own `mg_wakeup()`. Create a helper dummy UDP listener (on 127.0.0.1) that discards all incoming data, and a client connection to that listener. In `mg_wakeup()`, write a single byte into the client. UDP writes bypass iobufs, they `sendto()` directly. So that should work fine from a multiple threads, IMO.

mcottrell1, 2021-04-06:
thanks, that works.

closed 2021-04-06 by mcottrell1