cesanta/mongoose #1258: close_conn() crashes in _DEBUG-mode

gvanem, 2021-04-04:
Building and using Mongoose for MSVC-2019 in `_DEBUG`-mode (option `-MDd`)
gives me crash from `close_conn()`. Call-stack:
```
dump1090!close_conn(struct mg_connection * c = 0x05ba7858)+0x10
dump1090!mg_mgr_poll(struct mg_mgr * mgr = 0x00671464, int ms = 0)+0x31d
dump1090!mg_mgr_free(struct mg_mgr * mgr = 0x00671464)+0x41
dump1090!main(int argc = 2, char ** argv = 0x058008a0)+0x95d
...
```

It crashes at the `LIST_DELETE(struct mg_connection, &c->mgr->conns, c);` statement since `c->mgr == 0xDDDDDDD`.

AFAICS caused by:
``` 
 memset(c, 0, sizeof(*c));
 free(c);
``` 

Which is really:
``` 
 memset(c, 0, sizeof(*c));
 free(c);
 memset(c, 0xDD, sizeof(*c));
``` 
in `_DEBUG`-mode.

gvanem, 2021-04-09:
I have some trouble regenerating the crash now. It could have been a fault in my program that causes it.

But I'm curious about the:
```c
  memset(c, 0, sizeof(*c));
  free(c);
```
at end of `close_conn()`. What purpose does it serve? Some naive belief that such a mem-area should never be 
touched again?

Like I wrote, a freed memory-block in `_DEBUG` mode is filled with `0xDD...0xDD`. I.e. [**Dead Memory**](https://stackoverflow.com/questions/370195/when-and-why-will-a-compiler-initialise-memory-to-0xcd-0xdd-etc-on-malloc-fre); 1st answer. This should really be old news for any programmer.....

But prior to fixing the possible fault in my program, this helped a lot without quite understanding why:
```diff
--- a/mongoose.c 2021-04-03 09:18:30
+++ b/mongoose.c 2021-04-09 19:54:31
@@ -3237,7 +3237,7 @@
     }

     if (c->is_draining && c->send.len == 0) c->is_closing = 1;
-    if (c->is_closing) close_conn(c);
+    if (c->is_closing && c->mgr) close_conn(c);
   }
 }
```

cpq, 2021-04-14:
```c
  memset(c, 0, sizeof(*c));
  free(c);
```

Here, we hope that any code that holds `c` and uses that after `close_conn()`, would segfault.  Of course, the subsequent malloc() can make a stray `c` pointer pointing to some completely unrelated data.

`if (c->is_closing && c->mgr) close_conn(c);` -- the check for `c->mgr` should not be there.

cpq, 2021-04-29:
@gvanem closing this until more context is known - feel free to reopen if there is more data available.

closed 2021-04-29 by cpq