cesanta/mongoose #1260: Got WSAECONNRESET which leads to closing connection

depler, 2021-04-07:
File **captive-dns-server\main.c** contains such code (Windows version):

```
static void fn(struct mg_connection* c, int ev, void* ev_data, void* fn_data) {
    if (ev == MG_EV_READ) {
        struct mg_dns_rr rr;  // Parse first question, offset 12 is header size
        size_t n = mg_dns_parse_rr(c->recv.buf, c->recv.len, 12, true, &rr);
        if (n > 0) {
            char* buf = malloc(12 + n + sizeof(answer));
            struct mg_dns_header* h = (struct mg_dns_header*)buf;
```

So if I put breakpoint on  **char* buf = malloc(12 + n + sizeof(answer));** and then call in console **nslookup rutor.info 127.0.0.1** - this will lead to closing connection with WSAECONNRESET error. As a result we've got endless loop with 100% CPU load here in main (due to mgr->conns is NULL now):

```
  for (;;) mg_mgr_poll(&mgr, 1000);         // Event loop
```
Looks like a bug. Should we filter WSAECONNRESET in **mg_sock_failed**?

depler, 2021-04-07:
UPD. Yep, there is a windows specific bug: https://stackoverflow.com/questions/34242622/windows-udp-sockets-recvfrom-fails-with-error-10054

cpq, 2021-04-14:
@depler thank you.

Could you provide a diff for your suggested change please?

depler, 2021-04-14:
Well, it seems more complicated. As for fixing this specific behavior I did this (full code at https://github.com/depler/dns-filter/blob/master/src/dns-filter.c):

```
bool fix_udp_behavior(void* socket)
{
#if _WIN32
    BOOL bValue = FALSE;
    DWORD dwBytesReturned = 0;

    int r1 = WSAIoctl(socket, SIO_UDP_CONNRESET, &bValue, sizeof(bValue), NULL, 0, &dwBytesReturned, NULL, NULL);
    int r2 = WSAIoctl(socket, SIO_UDP_NETRESET, &bValue, sizeof(bValue), NULL, 0, &dwBytesReturned, NULL, NULL);
    return (r1 == 0 && r2 == 0);
#else
    return true;
#endif
}
```
This will prevent receiving errors **WSAECONNRESET** and **WSAENETRESET**, but obviously this is not enough. For example, if received UDP datagram size is more than MG_IO_SIZE (512 bytes by default) - we will get **WSAEMSGSIZE** and then... right, closing socket with endless loop. So I had to rewrite mg_mgr_poll special for handling UDP:

```
void mg_mgr_poll_udp(struct mg_mgr* mgr, int ms)
{
    mg_iotest(mgr, ms);

    struct mg_connection *connection, *next;
    for (connection = mgr->conns; connection != NULL; connection = next)
    {
        next = connection->next;
        mg_call(connection, MG_EV_POLL, NULL);

        if (connection->is_readable)
            read_conn(connection, ll_read);

        //always listen
        if (connection->is_closing)
            connection->is_closing = 0;
    }
}
```
This is not best solution, but I'm not ready to dig deeper and refactor socket read-write code. Probably, a lot of WSAEXX errors can be safely ignored to prevent socket closing (or may be just ignore all of them for UDP since it is connectionless protocol).

closed 2021-05-31 by depler