cesanta/mongoose #1263: About the https connection handshake failure when using openssl

GhostHallyz, 2021-04-14:
When i use mongoose to build https server, it always fails the handshake.

`
int mg_tls_handshake(struct mg_connection *c) {
  struct mg_tls *tls = (struct mg_tls *) c->tls;
  int rc;
  SSL_set_fd(tls->ssl, (long) c->fd);
  rc = c->is_client ? SSL_connect(tls->ssl) : SSL_accept(tls->ssl);
  if (rc == 1) {
    LOG(LL_DEBUG, ("%lu success", c->id));
    c->is_tls_hs = 0;
    return 1;
  } else {
    int code;
    ERR_print_errors_fp(stderr);
    code = mg_tls_err(tls, rc);
    if (code != 0) mg_error(c, "tls hs: rc %d, err %d", rc, code);
    return 0;
  }
}
`

In mongoose 7.2 version， at line 3690 of the mongoose.c file,  I found that ` rc = c->is_client ? SSL_connect(tls->ssl) : SSL_accept(tls->ssl);`  always return -1，it may crach when run to function ERR_print_errors_fp ().
Can I use `rc = SSL_accept(tls->ssl)` instead of `rc = c->is_client? SSL_connect(tls->ssl): SSL_accept(tls->ssl)`？

cpq, 2021-04-14:
Could you provide a diff for your suggested change please?

GhostHallyz, 2021-04-15:
An https client requests, SSL_accept fails for the first time, and succeeds for the second time.

cpq, 2021-04-29:
That's expected thogh.
Since mongoose is non-blocking, the first call to handshake might fail.
As TLS layer receives handshake data, it succeeds.

closed 2021-04-29 by cpq

GhostHallyz, 2021-05-07:
i see，thanks

hql420621, 2021-08-20:
i also run into this problem,is there a way to solve it,thanks