cesanta/mongoose #1267: mg_tls_init returns 0

Samega7Cattac, 2021-04-19:
hi,
I wanted to implement a http server with TLS and I checked the restful example to do it, but still `mg_tls_init` returns 0, which I don't know what it means, can't find any documentation of it, and `mg_tls_init(): 2021-04-19 13:15:13  E mongoose.c:410:mg_error   2 TLS is not enabled`.
```cpp
#include <iostream>

extern "C"
{
    #include "mongoose.h"
}

constexpr const char WWW[] = "./www";

constexpr const char URL[] = "https://127.0.0.1:8000";

constexpr const char SSI_PATTERN[] = "#.shtml";

constexpr const char CERT_PATH[] = "./cert.pem";

constexpr const char CERT_KEY_PATH[] = "./key.pem";

void handler(struct mg_connection *conn, int event, void *event_data, void *func_data)
{
    switch(event)
    {
        case MG_EV_ACCEPT:
            if (mg_url_is_ssl(URL))
            {
                std::cout << "[DEBUG] enabling TLS..." << std::endl;
                struct mg_tls_opts tls_opts = {
                        //.ca = "ca.pem",         // Uncomment to enable two-way SSL
                        .cert = CERT_PATH,     // Certificate PEM file
                        .certkey = CERT_KEY_PATH  // This pem conains both cert and key
                };
                std::cout << mg_tls_init(conn, &tls_opts) << " was returned by mg_tls_init()" << std::endl;
            }
            break;
        case MG_EV_HTTP_MSG:
            std::cout << "[DEBUG] serving page" << std::endl;
            struct mg_http_serve_opts opts = {WWW, SSI_PATTERN, nullptr};
            mg_http_serve_dir(conn, (mg_http_message*)event_data, &opts);
            break;
    }
}

int main() {
    struct mg_mgr webserver{};
    mg_mgr_init(&webserver);
    mg_log_set("3");
    mg_http_listen(&webserver, URL, handler, nullptr);
    for (;;) mg_mgr_poll(&webserver, 1000);
    mg_mgr_free(&webserver);
    return 0;
}
```
Tested with: `curl --cert ./cert.pem --key ./key.pem https://127.0.0.1:8000 -v --insecure`
Server output:
```
2021-04-19 13:20:15  I mongoose.c:3139:mg_listen 1 accepting on https://127.0.0.1:8000
2021-04-19 13:20:20    mongoose.c:3067:accept_co 2 accepted 127.0.0.1:60424
[DEBUG] enabling TLS...
2021-04-19 13:20:20  E mongoose.c:410:mg_error   2 TLS is not enabled
0 was returned by mg_tls_init()
2021-04-19 13:20:21    mongoose.c:2947:close_con 2 closed
```
Curl output:
```
*   Trying 127.0.0.1:8000...
* TCP_NODELAY set
* Connected to 127.0.0.1 (127.0.0.1) port 8000 (#0)
* ALPN, offering h2
* ALPN, offering http/1.1
* successfully set certificate verify locations:
*   CAfile: /etc/ssl/certs/ca-certificates.crt
  CApath: /etc/ssl/certs
* TLSv1.3 (OUT), TLS handshake, Client hello (1):
* OpenSSL SSL_connect: Connection reset by peer in connection to 127.0.0.1:8000 
* Closing connection 0
curl: (35) OpenSSL SSL_connect: Connection reset by peer in connection to 127.0.0.1:8000 
```

oamc, 2021-04-22:
I don't see your build CFLAGS, but looks to me like you did not declare any flags for TLS libraries.

https://cesanta.com/docs/#build-options

Samega7Cattac, 2021-04-28:
indeed it was missing `-lssl -lcrypto -DMG_ENABLE_OPENSSL` on my flags