cesanta/mongoose #1270: signal SIGPIPE, Broken pipe with mbedtls SSL on mg_tls_handshake.

djpiper28, 2021-04-22:
```gdb
#0  0x00007ffff7f7f7df in write () from /usr/lib/libpthread.so.0
#1  0x00007ffff7c1ca73 in mbedtls_net_send () from /usr/lib/libmbedtls.so.13
#2  0x00007ffff7c242c4 in mbedtls_ssl_flush_output () from /usr/lib/libmbedtls.so.13
#3  0x00007ffff7c24d1f in mbedtls_ssl_write_record () from /usr/lib/libmbedtls.so.13
#4  0x00007ffff7c2639e in mbedtls_ssl_write_handshake_msg () from /usr/lib/libmbedtls.so.13
#5  0x00007ffff7c31477 in mbedtls_ssl_write_certificate () from /usr/lib/libmbedtls.so.13
#6  0x00007ffff7c32b01 in mbedtls_ssl_handshake () from /usr/lib/libmbedtls.so.13
#7  0x000055555566c33e in mg_tls_handshake (c=0x7ffff0013720) at src/tls.c:37
#8  0x000055555566bab1 in mg_mgr_poll (mgr=0x7ffff7654dc0, ms=100) at src/sock.c:578
#9  0x000055555558b013 in pollingThread (apiIn=0x7fffffffdbb8) at buildtmp/apiserver.c:406
#10 0x00007ffff7f76299 in start_thread () from /usr/lib/libpthread.so.0
#11 0x00007ffff7774053 in clone () from /usr/lib/libc.so.6
```

does not occur very often but holding f5 in the browser will reproduce it in a few seconds.

Conditions that mg_tls_init is called.
```c
    if (event == MG_EV_ACCEPT) {
        //Init connection struct
        struct ServerConnection *s = (struct ServerConnection *)
            malloc (sizeof(struct ServerConnection));
        initServerConnection(s, api);
        
        c->fn_data = (void *) s;        
        
        if (mg_url_is_ssl(api->config.bindAddr)) {
            mg_tls_init(c, &api->opts);
        }
    }
```

djpiper28, 2021-04-22:
I am assuming this a bug in mbedtls not mongoose

cpq, 2021-04-29:
@djpiper28 Thank you.

The `mg_mgr_init` code ignores `SIGPIPE`:

```c
#elif defined(__unix) || defined(__unix__) || defined(__APPLE__)
  // Ignore SIGPIPE signal, so if client cancels the request, it
  // won't kill the whole process.
  signal(SIGPIPE, SIG_IGN);
#endif
```

What is your build environment? Maybe you don't have `__unix` or `__unix__` defined?

djpiper28, 2021-04-29:
Neither of those have been defined in my build environment

cpq, 2021-04-29:
@djpiper28 What is your build environment ?

djpiper28, 2021-04-29:
It was gnu make at the time of the issue but I've since changed to cmake and I don't know if the issue persists as I added the ignore SIGPIPE manually as a quick fix

cpq, 2021-04-29:
@djpiper28  Thank you!
The reason I am asking is to change the sources so others with the same environment would not have the same issue.

The information about the gnu make or cmake is irrelevant here.
What is relevant is the compiler and target OS. Could you reveal that please?

djpiper28, 2021-04-29:
Yeah so it g++ on manjaro linux

cpq, 2021-04-29:
Thank you.
Could you share the output of the following command please:

```sh
 g++ -dM -E -x c++ - < /dev/null
```

djpiper28, 2021-04-30:
```c
#define __SSP_STRONG__ 3
#define __DBL_MIN_EXP__ (-1021)
#define __cpp_attributes 200809L
#define __UINT_LEAST16_MAX__ 0xffff
#define __ATOMIC_ACQUIRE 2
#define __FLT128_MAX_10_EXP__ 4932
#define __FLT_MIN__ 1.17549435082228750796873653722224568e-38F
#define __GCC_IEC_559_COMPLEX 2
#define __cpp_aggregate_nsdmi 201304L
#define __UINT_LEAST8_TYPE__ unsigned char
#define __SIZEOF_FLOAT80__ 16
#define __INTMAX_C(c) c ## L
#define __CHAR_BIT__ 8
#define __UINT8_MAX__ 0xff
#define __SCHAR_WIDTH__ 8
#define __WINT_MAX__ 0xffffffffU
#define __FLT32_MIN_EXP__ (-125)
#define __cpp_static_assert 200410L
#define __ORDER_LITTLE_ENDIAN__ 1234
#define __SIZE_MAX__ 0xffffffffffffffffUL
#define __WCHAR_MAX__ 0x7fffffff
#define __GCC_HAVE_SYNC_COMPARE_AND_SWAP_1 1
#define __GCC_HAVE_SYNC_COMPARE_AND_SWAP_2 1
#define __GCC_HAVE_SYNC_COMPARE_AND_SWAP_4 1
#define __DBL_DENORM_MIN__ double(4.94065645841246544176568792868221372e-324L)
#define __GCC_HAVE_SYNC_COMPARE_AND_SWAP_8 1
#define __GCC_ATOMIC_CHAR_LOCK_FREE 2
#define __GCC_IEC_559 2
#define __FLT32X_DECIMAL_DIG__ 17
#define __FLT_EVAL_METHOD__ 0
#define __cpp_binary_literals 201304L
#define __FLT64_DECIMAL_DIG__ 17
#define __GCC_ATOMIC_CHAR32_T_LOCK_FREE 2
#define __cpp_variadic_templates 200704L
#define __UINT_FAST64_MAX__ 0xffffffffffffffffUL
#define __SIG_ATOMIC_TYPE__ int
#define __DBL_MIN_10_EXP__ (-307)
#define __FINITE_MATH_ONLY__ 0
#define __cpp_variable_templates 201304L
#define __FLT32X_MAX_EXP__ 1024
#define __FLT32_HAS_DENORM__ 1
#define __UINT_FAST8_MAX__ 0xff
#define __cpp_rvalue_reference 200610L
#define __FLT32_MAX_10_EXP__ 38
#define __DEC64_MAX_EXP__ 385
#define __INT8_C(c) c
#define __INT_LEAST8_WIDTH__ 8
#define __UINT_LEAST64_MAX__ 0xffffffffffffffffUL
#define __SHRT_MAX__ 0x7fff
#define __LDBL_MAX__ 1.18973149535723176502126385303097021e+4932L
#define __FLT64X_MAX_10_EXP__ 4932
#define __FLT64X_HAS_QUIET_NAN__ 1
#define __UINT_LEAST8_MAX__ 0xff
#define __GCC_ATOMIC_BOOL_LOCK_FREE 2
#define __FLT128_DENORM_MIN__ 6.47517511943802511092443895822764655e-4966F128
#define __UINTMAX_TYPE__ long unsigned int
#define __linux 1
#define __DEC32_EPSILON__ 1E-6DF
#define __FLT_EVAL_METHOD_TS_18661_3__ 0
#define __unix 1
#define __UINT32_MAX__ 0xffffffffU
#define __GXX_EXPERIMENTAL_CXX0X__ 1
#define __FLT128_MIN_EXP__ (-16381)
#define __WINT_MIN__ 0U
#define __FLT128_MIN_10_EXP__ (-4931)
#define __INT_LEAST16_WIDTH__ 16
#define __SCHAR_MAX__ 0x7f
#define __FLT128_MANT_DIG__ 113
#define __WCHAR_MIN__ (-__WCHAR_MAX__ - 1)
[... 303 lines cut ...]
#define __FLT32X_DIG__ 15
#define __PTRDIFF_TYPE__ long int
#define __ATOMIC_SEQ_CST 5
#define __FLT32X_MIN_10_EXP__ (-307)
#define __UINTPTR_TYPE__ long unsigned int
#define __DEC64_SUBNORMAL_MIN__ 0.000000000000001E-383DD
#define __DEC128_MANT_DIG__ 34
#define __LDBL_MIN_10_EXP__ (-4931)
#define __cpp_generic_lambdas 201304L
#define __SSE_MATH__ 1
#define __SIZEOF_LONG_LONG__ 8
#define __cpp_user_defined_literals 200809L
#define __FLT128_DECIMAL_DIG__ 36
#define __GCC_ATOMIC_LLONG_LOCK_FREE 2
#define __FLT32_HAS_QUIET_NAN__ 1
#define __FLT_DECIMAL_DIG__ 9
#define __UINT_FAST16_MAX__ 0xffffffffffffffffUL
#define __LDBL_NORM_MAX__ 1.18973149535723176502126385303097021e+4932L
#define __GCC_ATOMIC_SHORT_LOCK_FREE 2
#define __UINT_FAST8_TYPE__ unsigned char
#define _GNU_SOURCE 1
#define __cpp_init_captures 201304L
#define __ATOMIC_ACQ_REL 4
#define __ATOMIC_RELEASE 3
```

cpq, 2021-04-30:
Thank you!

Weird, I can see `__unix` defined in the output:

```c
#define __unix 1
```

Not sure how the `mg_mgr_init` SIGPIPE snippet was not triggered..

djpiper28, 2021-04-30:
I'll double check to see if my debugger throws all sigs

cpq, 2021-05-28:
@djpiper28 thanks for the report - although it is unclear what is the action here.
It looks like Mongoose's logic is fine, `SIGPIPE ` should have been triggered.
Closing this for now, feel free to reopen if some new facts are discovered.

closed 2021-05-28 by cpq

andrislusis, 2022-12-19:
I am letting you know that we are moving away from mongoose because of this bug (no SSL was used).

scaprile, 2022-12-19:
> I am letting you know that we are moving away from mongoose because of this bug (no SSL was used).

I personally would rather like that you open a new issue of your own, follow the template, and clearly explain what your problem is so we can help you. You are not the OP of this issue and there have been several new Mongoose releases since this issue was closed without the OP re-opening it.