cesanta/mongoose #1274: V7 series is not ready for production

quartzjer, 2021-04-29:
The porting effort to move from v6.x to v7 is *tremendous*, nearly every API and interface has changed, and there's no guide except to try compiling and fixing each error/warning.

The lwip option doesn't even compile, fundamental basic errors in the main source code such as this [mg_resolve](https://github.com/cesanta/mongoose/blob/16a7ba6bf0bad5086a815739a029a5bd7c4f7309/src/lwip.c#L150) call has the wrong args.

Support for setting the protocol on websocket connections has just disappeared, left up to the application to insert a custom header with it.

There's no ability to customize the TLS settings (the old _opt() interface) when creating a connect.

Lost the ability to set custom user flags on connections.

I can go on, but it's not worth it.

The good part: the V7 code and library interface is **significantly** better... it's just not production ready yet.

cpq, 2021-04-29:
@quartzjer thanks.

Your comments are biased by the 6.x API - but of course appreciated.

From the point of view of the new user, with no prior 6.x experience, the only issue I can see is the TLS options and the low level LWIP support - which probably should be dropped anyway.

Other than the TLS opts and low level LWIP, anything else to mention?

lnksz, 2021-04-29:
SocksV5 was "factored out" from the lib. The [example](https://github.com/cesanta/mongoose/blob/master/examples/socks5-server/main.c) is then a suggested implementation.

Support for HTTP digest authentication was also dropped.

Unfortunately, these changes are not mentioned in the [v7+ release notes](https://github.com/cesanta/mongoose/releases/tag/7.0).

cpq, 2021-04-29:
@lnksz thanks, release notes updated.

quartzjer, 2021-04-29:
> the only issue I can see is the TLS options and the low level LWIP support - which probably should be dropped anyway.

You're considering dropping LwIP support?

> Other than the TLS opts and low level LWIP, anything else to mention?

When the release source code for 7.2 doesn't even build without basic compile errors in a core feature (LWIP) I have a very hard time considering the project production ready.

cpq, 2021-04-29:
@quartzjer thank you.

LWIP provides several APIs: low-level, "custom lwip", and BSD.
There are many projects that use Mongoose with BSD LWIP API.  That API is implemented by `src/api/sockets.c` inside LWIP source tree and is enabled by `LWIP_SOCKET` option.

I said "dropping low level LWIP support", I guess you missed the "low level" part and think and Mongoose would not build with any project that uses LWIP. That is a wrong assumption. There are many projects that use LWIP and 7.2 and are in production currently.

My other guess is that your project uses a low-level LWIP API, not a BSD one. The low level LWIP API support is currently broken in 7.2, making you state that 7.2 is not production ready - which is also wrong. What is right is that 7.2 low level support is broken.

Could you confirm that both my assumptions are correct, please?
Also, could you reveal what is your build environment - compiler, SDK, and libraries used, please?

cpq, 2021-05-01:
@quartzjer Thank you!

Your environment allows for BSD sockets API in LWIP.  
What is your Mongoose build flags? If you have enabled `MG_ENABLE_LWIP`, could you drop it and rebuild, please?

See https://cesanta.com/docs/#custom-build to add any custom headers if required - but don't disable `MG_ENABLE_SOCKET ` like in the doc.

quartzjer, 2021-05-04:
It required adding `#include <socket.h>` to `mongoose.h` and in `mongoose.c` at the bottom `mg_millis()` there is no STM32 support so had to add:
```
#elif MG_ARCH == MG_ARCH_STM32
  return HAL_GetTick();
```

And using the latest IAR 9 with `MG_ENABLE_MBEDTLS` enabled it fails to build with an error due to using unsupported new C features and all these warnings:

```
 Warning[Pe550]: variable "h" was set but never used Middlewares\Third_Party\Mongoose\mongoose.c 198 
 Warning[Pa181]: incompatible redefinition of macro "closesocket" (defined Hat line 628 of "EWARM/../Middlewares/Third_Party/LwIP/src/include\lwip/sockets.h"h) Middlewares\Third_Party\Mongoose\mongoose.c 2614 
 Warning[Pe550]: variable "rc" was set but never used Middlewares\Third_Party\Mongoose\mongoose.c 3080 
 Warning[Pa134]: left and right operands are identical Middlewares\Third_Party\Mongoose\mongoose.c 3116 
 Warning[Pe550]: parameter "n" was set but never used Middlewares\Third_Party\Mongoose\mongoose.c 3448 
 Warning[Pe223]: function "mbedtls_x509_crt_parse_file" declared implicitly Middlewares\Third_Party\Mongoose\mongoose.c 3499 
 Error[Pe060]: this operator is not allowed in an integral constant expression Middlewares\Third_Party\Mongoose\mongoose.c 3507 
 Error[Pe060]: this operator is not allowed in an integral constant expression Middlewares\Third_Party\Mongoose\mongoose.c 3507 
 Warning[Pe223]: function "mbedtls_x509_crt_parse_file" declared implicitly Middlewares\Third_Party\Mongoose\mongoose.c 3518 
 Warning[Pe223]: function "mbedtls_pk_parse_keyfile" declared implicitly Middlewares\Third_Party\Mongoose\mongoose.c 3522 
 Warning[Pe223]: function "gettimeofday" declared implicitly Middlewares\Third_Party\Mongoose\mongoose.c 4157 
 Warning[Pe223]: function "usleep" declared implicitly Middlewares\Third_Party\Mongoose\mongoose.c 4168 
```

cpq, 2021-05-12:
@quartzjer Thank you.

Apparently, the right way to tackle this is to have those particular environments tested via CI.
We've done the first step - added a bare minimum example/stm32-freertos-tcp.
It uses:
- GCC toolchain
- hand-written headers / bootstrap for STM32
- [FreeRTOS/FreeRTOS-Kernel](https://github.com/FreeRTOS/FreeRTOS-Kernel) V10.4.3
- [FreeRTOS/FreeRTOS-Plus-TCP](https://github.com/FreeRTOS/FreeRTOS-Plus-TCP) V2.3.2 (freertos's own TCP stack)

The setup does not use a real hardware, only stubs, so networking-wise, all it tests is a clean build.
However it builds cleanly, and runs two FreeRTOS tasks - one blinks and LED, another serves HTTP.

The next step is to add the same thing for LWIP: STM32 + FreeRTOS + LWIP (using BSD sockets API)

The next step is to add mbedTLS to the test.

Then, ultimately, hook up real boards and reflash / test on a real hardware in a lab, triggering over GH CI.

So we're on the FreeRTOS + LWIP test step - hopefully that will clear off the issues you have.

Of course your feedback is welcome - please share if you have any comments.

quartzjer, 2021-05-12:
This is great progress, thank you @cpq !

cpq, 2021-09-04:
Now we have several examples covering several boards. Closing this.
Thanks for your attention and patience.

closed 2021-09-04 by cpq