cesanta/mongoose #1281: Connection fails when DNS returns more than 5 address for the specified URL

GlebPlekhotko, 2021-05-11:
Function _mg_dns_parse_ in the _mongoose.c_ file has this sanity check:
`if (mg_ntohs(h->num_answers) > 5) return 0; `

Recently I've got a case when DNS reported 8 possible IP address for a given URL, so sanity check failed and connection attempt thereon. I fixed in a most straightforward way:

In _mongoose.c_ file

`if (mg_ntohs(h->num_answers) > 10) return 0;    // Sanity`

In _mongoose.h_ file:
```
struct mg_dns_message {
uint16_t txnid;   // Transaction ID
char name[512];   // Host name 
uint32_t ipaddr;  // Resolved IPv4 address
};
```

But maybe it is better to just pick up the very first IP address and ignore the rest?

cpq, 2021-05-28:
The current source already has a limit of 10:

https://github.com/cesanta/mongoose/blob/master/src/dns.c#L101-L104

What version do you work with?

GlebPlekhotko, 2021-05-28:
Yes I see. 
I'm not sure, but when I published this issue the latest version did have limit of 5. But maybe I messed up something.
Please close and thanks for reply.

closed 2021-05-29 by cpq