cesanta/mongoose #1285: format-truncation error on arm 32 bit plattform

jcorporation, 2021-05-12:
Code snippet from mongoose.c:
```
        snprintf(range, sizeof(range),
                 "Content-Range: bytes " MG_INT64_FMT "-" MG_INT64_FMT
                 "/" MG_INT64_FMT "\r\n",
                 r1, r1 + cl - 1, (int64_t) st.st_size);
```
Line 963 is the line with Content-Range

Compiler error:
```
/data/src/myMPD/dist/src/mongoose/mongoose.c: In function ‘mg_http_serve_file’:
/data/src/myMPD/dist/src/mongoose/mongoose.c:963:18: error: ‘%lld’ directive output may be truncated writing between 1 and 11 bytes into a region of size between 8 and 45 [-Werror=format-truncation=]
                  "Content-Range: bytes " MG_INT64_FMT "-" MG_INT64_FMT
                  ^~~~~~~~~~~~~~~~~~~~~~~
In file included from /data/src/myMPD/dist/src/mongoose/mongoose.c:18:
/data/src/myMPD/dist/src/mongoose/mongoose.h:266:23: note: format string is defined here
 #define MG_INT64_FMT "%" PRId64
/data/src/myMPD/dist/src/mongoose/mongoose.c:963:18: note: directive argument in the range [-2147483648, 2147483647]
                  "Content-Range: bytes " MG_INT64_FMT "-" MG_INT64_FMT
                  ^~~~~~~~~~~~~~~~~~~~~~~
In file included from /usr/include/stdio.h:873,
                 from /data/src/myMPD/dist/src/mongoose/mongoose.h:253,
                 from /data/src/myMPD/dist/src/mongoose/mongoose.c:18:
/usr/include/arm-linux-gnueabihf/bits/stdio2.h:67:10: note: ‘__builtin___snprintf_chk’ output between 29 and 76 bytes into a destination of size 70
   return __builtin___snprintf_chk (__s, __n, __USE_FORTIFY_LEVEL - 1,
          ^~~~~~~~~~~~~~~~~~~~~~~~~~~~~~~~~~~~~~~~~~~~~~~~~~~~~~~~~~~~
        __bos (__s), __fmt, __va_arg_pack ());
        ~~~~~~~~~~~~~~~~~~~~~~~~~~~~~~~~~~~~~
cc1: all warnings being treated as errors
make[2]: *** [CMakeFiles/mympd.dir/build.make:752: CMakeFiles/mympd.dir/dist/src/mongoose/mongoose.c.o] Error 1
make[1]: *** [CMakeFiles/Makefile2:96: CMakeFiles/mympd.dir/all] Error 2
make: *** [Makefile:130: all] Error 2

```

cpq, 2021-05-27:
What's the toolchain, is it `arm-none-eabi` ?
Please dump all compilation flags.

jcorporation, 2021-05-27:
Target: arm-linux-gnueabihf
gcc version 8.3.0 (Raspbian 8.3.0-6+rpi1) 

```
/usr/bin/cc  -I/usr/include/lua5.3 -I/data/src/myMPD/release -I/data/src/myMPD -I/data/src/myMPD/dist/src/libmpdclient/include  -std=gnu11 -O3 -Wall -Werror -Wvla -Winit-self -Wuninitialized -Werror=array-bounds -fstack-protector-strong -D_FORTIFY_SOURCE=2 -pie -fPIE -O3 -DNDEBUG -s   -Wno-unused-result -Wno-vla -DMG_ENABLE_OPENSSL=1 -DMG_ENABLE_DIRECTORY_LISTING=1 -o CMakeFiles/mympd.dir/dist/src/mongoose/mongoose.c.o   -c /data/src/myMPD/dist/src/mongoose/mongoose.c

```

cpq, 2021-05-28:
Thank you Jürgen.
We'll add an extra ARM test target with the `-O3` flag that actually triggers that warning.
We'll change default test flags too, to make them as strict as possible.

commit 736b8ad7 (2021-05-28): Fix #1285 - __builtin___snprintf_chk warning. Tighten test flags and refactor the code to squash all warnings
closed 2021-05-28 by cpq