cesanta/mongoose #1288: mg_iobuf_append allocates too much memory

GlebPlekhotko, 2021-05-21:
I'm using websocket connection and overall size of the frame is limited to 4K. It is constituted of 8 bytes for the header (1 byte opcode + 1 byte length equal to 126 + 2 bytes for the extended length + 4 masking key) and 4088 bytes of payload. itself. _MG_IO_SIZE_ is equal to 2048 bytes so I expect exactly 4K of the dynamic memory to be allocated. 

Instead it allocates 6K despite all the data fits into first 4K. Here is how it goes:

1. _mg_ws_send_ is called, _len_ argument is 4088;
2. it sees payload length to be inside (126, 65536) range and so sets _payload len_ byte to 126 and puts actual length to 2 bytes long extended payload, making header 4 bytes long;
3. it calls _mg_send_ first time and it calls _mg_io_buf_append_;
4. no memory is allocated at all at this moment, so 2K (_MG_IO_SIZE_) is obtained from the heap;
5. 4 bytes of the header is copied there;
6. control returns to _mg_ws_send_ and it performs the same sequence for the masking key, it is also 4 bytes long and fits into allocated before 2K bytes chunk;
7. at this moment _c->send.len_ is 8 and _c->send.size_ is 2048;
8. now it's payload's turn, _mg_send_ again invokes _mg_io_buf_append_, _len_ argument is 4088, _chunk_size is 2048;
9. _new_size_ in this case is determined to be 6144 bytes long and exactly this amount of memory is allocated;
10. header's and payload's data is concatenated in the new buffer, old one is freed;
11. control returns to _mg_ws_send_;
12.  _c->send.len_ is 4096 and _c->send.size_ is 6144;

This may seem to be not a very big issue at glance, but in resource-constrained applications, particularly microcontroller based, it is not welcomed to use more resource than you actually need.

cpq, 2021-05-22:
Before reading into the `c->recv` buffer, Mongoose makes sure that it has space which is `MG_IO_SIZE` aligned:
- If `MG_IO_SIZE` is 2048 and `c->recv.len == 0`, then `c->recv.size` is resized to 2048
- If `MG_IO_SIZE` is 2048 and `c->recv.len == 1`, then `c->recv.size` is resized to 4096, because free space is less than `MG_IO_SIZE`, and allocation granularity is `MG_IO_SIZE`.

So what you see is by design. Make `MG_IO_SIZE` smaller to tighten RAM usage.

Said that, what allocation strategy you'd expect?

GlebPlekhotko, 2021-05-24:
That's about receive. 

I'm talking about sending data. I still have one byte free in previously allocated buffer and I it would be fine to utilize it. Instead this strategy allocates new chunk of memory but does not use it. 

Yes, I can partially alleviate it by decreasing _MG_IO_SIZE_, but in my case that rises some limitations, because I don't have full-fledged sockets layer. It is actually running in hardware. If I specify read size to be 256 bytes, but 257 bytes arrives, the very first byte becomes overwritten. :-) Hence read size must be so to obtains complete websocket frame of maximal size in one shot.

cpq, 2021-05-27:
Thanks for sharing!

The reason to chunk data (both incoming and outgoing) is to increase performance and decrease RAM fragmentation - at the expense of suboptimal RAM usage. Currently, the only tunable is `MG_IO_SIZE`. It its extreme, it could be set to 1, which gives the best RAM usage - but awful performance (many tiny 1-byte reads and reallocations) and RAM fragmentation.

When you read, mongoose makes sure you have at least `MG_IO_SIZE` space in the `c->recv`, and makes it aligned to decrease fragmentation.

When you write, mongoose copies your data to `c->send` and again, makes it `MG_IO_SIZE` aligned to decrease fragmentation.

What we can do is to keep the `MG_IO_SIZE` buffering for reads, but turn it off for writes. Is that what you're looking for?

GlebPlekhotko, 2021-05-28:
Of course no! :-) I just propose to slightly change condition on which _mg_io_append_ decides to add one more chunk. Let's consider that on the following examples.

Preconditions:
1. MG_IO_SIZE is 2048;
2. io->size is 2048;
3. io->len = 2047;
4. _mg_iobuf_append_ is called, _len_ is 1;

Current flow for the _**new_size**_ variable:

_size_t **new_size** = io->len + len + chunk_size;_
_**new_size** -= new_size % chunk_size;_
_if (**new_size** != io->size) mg_iobuf_resize(io, **new_size**);_

Being calculated for the aforementioned preconditions __**new_size**__ amounts to 4096, so _io->buf_ gets reallocated.

We may use something like this instead:

_size_t **new_size** = io->len + len;_
_if (**new_size** > io->size) {_
_    size_t align = new_size % chunk_size;_
_    if (align) align = chunk_size - align;_
_    **new_size** += align;_
_    mg_iobuf_resize(io, **new_size**);_
_    if (**new_size** != io->size) len = 0;  // Realloc failure, append nothing_
_}_

That gives us for the same preconditions __**new_size**__ amounting to 2048, so _io->buf_ remains the same. At the same time your requirement to align all future allocations to MG_IO_SIZE is preserved. Of course there is some overhead, in comparison to original approach, but from my standpoint its negligible.

commit 9d68e20d (2021-05-29): Fix #1288 - better mg_iobuf_append allocation strategy
closed 2021-05-29 by cpq